pybricks/support #64: [Bug] Pybricksdev Flash command don't work on Windows 10

GianCann, 2020-06-21:
S.O. Windows 10 64 Bit

The first problem is in the Bleak library.

```
            devices = await scanner.get_discovered_devices()
            for dev in devices:
                print(dev.name)
                # If the name matches, stop scanning and return address.
                if name == dev.name:
                    await scanner.stop()
                    return dev.address
```

dev.name return always 'Unknown' as the name of device discovered.
So, I haved tried to modify the code in the ble.py to use the MAC Address (that I have got with another piece of code):

```
            devices = await scanner.get_discovered_devices()
            for dev in devices:
                print(dev.address)
                #if dev.address=="90:84:2B:2C:93:A9":
                #    print("LEGO HUB")
                
                # If the name matches, stop scanning and return address.
                if dev.address=="90:84:2B:2C:93:A9":
                    print("Device Found")
                    await scanner.stop()
                    return dev.address
```
Now, the flash procedure go into the next step (connection) but without success:

```
C:\Users\Gianluca\pybricksdev>poetry run pybricksdev flash cplushub-firmware.zip
Creating firmware
Searching for LEGO Bootloader
Device discovered
D8:8A:DC:97:11:AF
90:84:2B:2C:93:A9
Device Found
Found: 90:84:2B:2C:93:A9
Connecting to 90:84:2B:2C:93:A9

```
The procedure remaining in the 'Connecting to...' status.
The Hub don't change his LED from purple blink to the Red/Green/Blue sequence.

GianCann, 2020-07-01:
I can confirm that with removing the callback function for server disconnection I able to flash the CityHub from Windows!
https://github.com/pybricks/pybricksdev/issues/2#issuecomment-652441168

laurensvalk, 2020-10-07:
> I can confirm that with removing the callback function for server disconnection I able to flash the CityHub from Windows!

Then we can close this. Thank you!

closed 2020-10-07 by laurensvalk