pybricks/support #65: [Suggestion] Change the script in main.py

GianCann, 2020-06-23:
Actually, the default code in main.py (in the xxx-firmware.zip file) contains:

```
motor = Motor(Port.B)
motor.run_angle(1000, 90)
```

This code cannot work now. See #41 

I suggest to change with:
```
import uos
print(uos.uname())
```

dlech, 2020-06-23:
We were actually thinking about printing something like that when Bluetooth is connected.

If that is the main program and Bluetooth is not connected, then running the program will seem like it does nothing. Maybe it would be better to blink the light or something like that in the default main program?

GianCann, 2020-06-23:
> Maybe it would be better to blink the light or something like that in the default main program?

Yes, is a good idea.
If no custom program is present, is useful to have a simple feedback "All it's ok" by Pybricks.
- BLE Connection present? LED blink + message
- BLE Connection not present? Only LED blink

laurensvalk, 2020-10-09:
This has been resolved for now. Thanks for the pull request!

closed 2020-10-09 by laurensvalk