pybricks/support #66: [Bug] Program will not start

XilefTech, 2020-06-23:
**Describe the bug**
After everything worked for quite some time now, I just wanted to upload my program once again to my brick. after the upload, it printed into the console:
`----------
/usr/bin/env: 'pybricks-micropython\r': No such file or directory
----------
Exited with error code 127.`

**To Reproduce**
Steps to reproduce the behavior:
1. use pybricks for a while
2. Click on upload and run, after not changing anything
3. See error
_I don't know, wether this is reproducable or not. I didn't change anything on the brick or in the program, that might have caused that behaviour myself._

**Expected behavior**
The program starting like it normally would

**Screenshots**
_nothing to screenshot here_

Anyone got an Idea what that could have caused that?
And how do I fix it? Do I have to rewrite the image on the SD-Card?

Thanks for helping in advance!

dlech, 2020-06-23:
The `\r` tells us that the file has Windows line endings. The ev3dev-browser VS Code extension should take care of this automatically, but you have to make sure the first line of the program is exactly:

    #!/usr/bin/env pybricks-micropython

With no spaces or hidden characters before, above, after, in-between, etc.

If you are not using the ev3dev-browser VS Code extension, then make sure to save the file with UNIX line endings instead of Windows line endings.

XilefTech, 2020-06-23:
[image]
that is exactly how it is written in my code. I use the ev3dev-browser VSCode extension and even tried copy-pasting the line, that you sent - nothing changed...

dlech, 2020-06-23:
Did you try changing the line endings from CRLF to LF?

[image]

[image]

XilefTech, 2020-06-23:
not yet, but as I stated before, everything worked as it was for moths now...

but I sure can try that

XilefTech, 2020-06-23:
now everything works again...
weird....
I never changed something like that or even knew, there was a difference like that.

Thanks for helping again!

closed 2020-06-23 by XilefTech