pybricks/support #67: [Feature] More control over the Brick-LEDs

XilefTech, 2020-06-23:
**Is your feature request related to a problem? Please describe.**
Nope, just a wish

**Describe the solution you'd like**
I would like to have more control over the brick-LEDs on the EV3. Rn, as far as the docs go, you can only turn them of or on, in one color. I think it would be cool, if you had like control of the actual brightnessof the individual colored LEDs. Also it would be nice, if you could control the right and the left ones separately. Maybe also a sine-wave-pulsing predefined option would be nice.

**Describe alternatives you've considered**
why alternatives? just something that would be cool, like proposed

**Additional context**
just thought, I could make some cool-looking effects with that on my button-LEDs. Maybe a mod to switch the yellow with blue would be great to have actual RGB in your brick! _HYPE_

dlech, 2021-02-05:
Also see #251

dbfin, 2021-02-05:
This would be a cool feature! 

- More predefined colors
- Independent sides
- Support for the built-in triggers (animations that you should not define yourself as they are built in ev3dev)
- And possibly the ability to define your own 2048 millisecond animations as in that function

But this will probably require some time and effort.

On the other hand, as the first step, in fact, if #251 is resolved (and now thanks to @dlech it seems a one-line change solves it), one would already be able to create whatever colors/animation they wish on both sides independently. You would be able to manipulate LEDs directly as described here: https://www.ev3dev.org/docs/tutorials/using-ev3-leds/. Again, later some official support for more features like above would be nice as well, especially the ability to utilize that function called every 32 milliseconds for your own animations!