pybricks/support #68: [Bug] DriveBase's drive command has an incorrect parameter name "drive_speed"

markovian72, 2020-06-26:
**Describe the bug**
Using The EV3's **DriveBase** library's **drive** command's tooltip shows "**drive_speed**" as a parameter name. An error is generated upon running the script, stating that the parameter is actually called "**speed**". Changing the parameter name to "speed" allows the script to run correctly.

I installed pybricks-stubs 2.0.2 to get intellisense auto-completions on my Manjaro installation.

**To Reproduce**
Steps to reproduce the behavior:
1. Create a DriveBase object:
`tank = DriveBase(left_motor=left_motor, right_motor=right_motor, wheel_diameter=32.6, axle_track=142)`
2. Execute a drive command with a parameter named "drive_speed":
`tank.drive(drive_speed=500, turn_rate=25)`
3. An exception is generated at the drive line number:
`Starting: brickrun --directory="/home/robot/snatch3r_test1" "/home/robot/snatch3r_test1/main.py"
----------
Traceback (most recent call last):
  File "/home/robot/snatch3r_test1/main.py", line 30, in <module>
TypeError: 'speed' argument required
----------
Exited with error code 1.`

**Screenshots**
[image]

dlech, 2020-06-26:
Also reported at https://github.com/pybricks/pybricks-api/issues/23

laurensvalk, 2020-07-07:
This is fixed in https://github.com/pybricks/pybricks-api/commit/82b16dc8de8064baa1af9ca8e7adf70081f9469e. Thank you for reporting!

closed 2020-07-07 by laurensvalk