pybricks/support #72: [Report] Flashing CityHub successful

GianCann, 2020-06-29:
After some problems in setting up the work environment, I managed to compile and load the firmware for the CityHub.

The working environment is Ubuntu 18.04, running in Virtualbox on Windows 10.

I noticed that pairing with code.pybricks.com does not happen if the Hub has a connected device.
You have to do the pairing first, and then connect the motor.

laurensvalk, 2020-06-29:
Please post all new issues in `support`. We reserve the other issue trackers for actual bug reports that need to get fixed.

That makes it easier for us to know what to fix.

Thanks!

laurensvalk, 2020-06-29:
> I noticed that pairing with code.pybricks.com does not happen if the Hub has a connected device.
You have to do the pairing first, and then connect the motor.

If you find this to happen consistently, please do open an issue for that as well. Nice find!

closed 2020-07-03 by dlech