pybricks/support #74: [Bug] Bluetooth status

GianCann, 2020-06-30:
**Describe the bug**
The Bluetooth button on "Pybricks Code" lost the actual status

**To reproduce**
Steps to reproduce the behavior:
1. Go to code.pybricks.com
2. Click on Bluetooth button
3. Connect with smart hub
4. Run some code...
5. Now, refresh the page
6. the Bluetooth button is now in the "not connected" status.
7. If you click again on the button, you see the hub already paired.

[image]

8. You have to click to again on "Pairing" (Accoppia, in Italian) to get the hub works.

**Expected behavior**
Maintaining the Bluetooth "connection" status, across refreshing the page

laurensvalk, 2020-06-30:
It is expected that refreshing causes the hub to disconnect. Think of it like restarting the Powered Up app - then it disconnects too.

Please note that paired does not mean connected. Paired just means it is in your computer's "address book", usually because you've connected to it before.

GianCann, 2020-06-30:
Thank you @laurensvalk  for the clarification

closed 2020-06-30 by laurensvalk