pybricks/support #75: add send() method to connection.py

PhoBricks, 2020-07-01:
**Is your feature request related to a problem? Please describe.**
It's a missing feature

**Describe the solution you'd like**
With this method we are able to send scratch.based  commands to the SPIKE Hub

**Describe alternatives you've considered**
-

**Additional context**
This could be a code example (done by I. Oba not me)

# Add send_command method into SpikePrimeStreamReader class
def send_command(self, command):
    self.sock.send((command + '\r').encode())

SpikePrimeStreamReader.send_command = send_command

laurensvalk, 2020-07-01:
Thanks for this suggestion and others. Would you be interested in making the proposed change yourself? We'll be happy to merge it.

If you're not familiar with Git, you can still do it using the online interface:

**- Create a fork**
This copies the projects to your account (top right button).
[image]

**- Modify the files**
In your copy, you can make the changes (click the pencil button).
[image]

You can enter a description of the changes. For inspiration, you can look at the other [commit messages](https://github.com/pybricks/pybricks-projects/commits/d5ee42156d47ae4aea5afed9d98b9a6c9de3f418).

**- Create a pull request**
When you're happy with the changes, make a pull request:
[image]

laurensvalk, 2020-07-18:
@PhoBricks wrote in #77:

> **Is your feature request related to a problem? Please describe.**
> SPIKE sends more outputs as only if m has the value 0. I discovered the following which would be great to be integrated into the connection.py too:
> 
> {"m":3,"p":["center", 170]}]}
> First value: Which Button is pressed on the SPIKE (center, leftside, rightside)
> Second value: Duration in ms how long the Button is pressed.
> 
> Gesture detection:
> {"m":4,"p":"tapped"}
> 
> Values:
> tapped
> doubletapped
> shake
> Front
> Rightside
> Leftside
> Up
> down
> 
> There may be more values...
> 
> **Describe the solution you'd like**
> Integrate in connection.py
> 
> ## **Describe alternatives you've considered**
> ## **Additional context**

laurensvalk, 2020-07-18:
@PhoBricks wrote in #76:

> **Is your feature request related to a problem? Please describe.**
> If a program is running on the SPIKE, the output uses also the last two values, it would be great to integrate them also into the connection.py
> 
> **Describe the solution you'd like**
> Example Output:
> {"m":0,"p":[[0, []], [0, []], [0, []], [0, []], [0, []], [0, []], [-9, -4, 988], [-1, 5, 0], [-1, 0, 0], "09090:99999:99999:09990:00900", 5005]}}
> Actual Displayed Image->"09090:99999:99999:09990:00900"
> Runtime in ms since program start -> 5005
> 
> ## **Describe alternatives you've considered**
> ## **Additional context**

closed 2021-07-13 by laurensvalk