pybricks/support #77: Process more outputs from SPIKE

PhoBricks, 2020-07-01:
**Is your feature request related to a problem? Please describe.**
SPIKE sends more outputs as only if m has the value 0. I discovered the following which would be great to be integrated into the connection.py too:

{"m":3,"p":["center", 170]}]}
First value:  Which Button is pressed on the SPIKE (center, leftside,  rightside)
Second value: Duration in ms how long the Button is pressed.

Gesture detection:
{"m":4,"p":"tapped"}

Values:
tapped
doubletapped
shake
Front
Rightside
Leftside
Up
down

There may be more values...

**Describe the solution you'd like**
Integrate in connection.py

**Describe alternatives you've considered**
-

**Additional context**
-

dlech, 2020-07-18:
Which `connection.py` are you referring to?

laurensvalk, 2020-07-18:
Closing and merging this with #75.

closed 2020-07-18 by laurensvalk