pybricks/support #78: [Bug] [pybricksdev] Stop program

GianCann, 2020-07-01:
**Describe the bug**
I tried to run a program from command line with:
poetry run pybricksdev run "(ble address") myscript.py

All works fine...

The myscript.py file contains an infinite loop (read colour sensor and print the name, with a 500ms delay from read to read)

When I press CTRL+C from the keyboard this stop the pybricksdev tool, but not the program running on the hub.

**Expected behavior**
Stop also the hub program

dlech, 2020-07-01:
Windows?

GianCann, 2020-07-01:
Yes, @dlech, I have this bad habit ... :smiley:

laurensvalk, 2020-07-01:
That’s not a bug - it’s the same as closing Pybricks Code in the browser. Your script keeps running.

The fact that programs keep running after becoming disconnected is one of the primary features of Pybricks 😉 

But we can certainly add a stop command equivalent to the stop button in Pybricks Code.

GianCann, 2020-07-01:
@laurensvalk what you say makes sense.
By the way, I was obviously able to connect to the hub from another device, and continue to see the output of the program running on the hub.

closed 2020-07-07 by GianCann