pybricks/support #81: [Question] Doesn't pybricks have sys.path that can be extended by sys.path.append(...)?

TheVinhLuong102, 2020-07-03:
I've factored certain reusable classes/funcs out to common utilities and put them under `/home/robot/utils` on my EV3Dev. 

However, when using pybricks, I am unable to extend the search path by `sys.path.append('/home/robot')` because of `AttributeError: 'module' object has no attribute 'path'`.

May I check if pybricks's `sys` module has any limitations that I need to be aware of? Thank you.

laurensvalk, 2020-07-03:
MicroPython does not have all the features of Python. See [sys](https://docs.micropython.org/en/latest/library/sys.html).

dlech, 2020-07-03:
>    sys.path.append('/home/robot')

This works for me using Pybricks 2.0.0 on EV3.

```python
>>> import sys
>>> dir(sys)
['__class__', '__name__', 'argv', 'byteorder', 'exit', 'getsizeof', 'implementation', 'maxsize', 'modules', 'path', 'platform', 'print_exception', 'stderr', 'stdin', 'stdout', 'version', 'version_info']
>>> sys.path
['', '', '/home/robot/.pybricks-micropython/lib', '/usr/lib/pybricks-micropython']
>>> sys.path.append('/home/robot')
>>> sys.path
['', '', '/home/robot/.pybricks-micropython/lib', '/usr/lib/pybricks-micropython', '/home/robot']
>>> 
```

Can you share a full working example that causes the problem?

TheVinhLuong102, 2020-07-04:
@dlech @laurensvalk sorry, my bad, I've found that the error in my code comes from the lack of `os.path` instead of `sys.path`. Let me close this issue then.

closed 2020-07-04 by TheVinhLuong102