pybricks/support #82: [Question] Motor speed value missing?

GianCann, 2020-07-03:
When I using motor.dc(70) and after I want to check del speed with motor.speed() the value returned is 0

It's a "by design" behavior?

GianCann, 2020-07-03:
Ok, I check that only less of 20% of duty the speed value is 0

dlech, 2020-07-03:
The actual speed of the motor should be returned. So if the motor is not actually moving, the returned speed will be 0.

GianCann, 2020-07-03:
DC Value:10 Speed value:0
DC Value:20 Speed value:0
DC Value:30 Speed value:210

dlech, 2020-07-03:
It takes a certain amount of power to overcome the internal friction of the motor before it starts turning. So this looks exactly as expected if nothing is connected to the motor.

If you block the motor from turning it will still have speed of 0 even at 100% duty cycle because it is not turning.

GianCann, 2020-07-03:
Ok, but with 20% of duty, the motor turn and so, I expect that return a speed value.

dlech, 2020-07-03:
Assuming this is not a built-in motor on a Move hub.

We would have to check the source code to be sure, but I think we are just reporting the speed sent via UART by the motor. So this is probably a question for LEGO. :smile:

laurensvalk, 2020-07-04:
Which motor are you using? Can you try doing the same with the powered up app?

GianCann, 2020-07-04:
I using the Boost Motor and the LEGO App return data about rotation starting from 15% of speed:
[image]

laurensvalk, 2020-07-04:
Thanks, but you’ll want to compare with dc, which they probably call power.

And rotation angle should be reported regardless of the speed. That is just not related at all. Didn’t you mention the speed measurement in the initial post? It would probably make sense to compare that to the speed block then, not the angle block.

GianCann, 2020-07-04:
This is the result of this code:

```
delay=1000 #First run: 1000, second run: 3000
while (x <= 50):
    x +=1
    mot.dc(x)
    print("a) DC Value:" + str(x) + " Speed value:" + str(mot.speed()))
    wait(delay)
    print("b) DC Value:" + str(x) + " Speed value:" + str(mot.speed()))
    wait(100)
```

I ran the code two time. The first with a delay of one secondo between the reading of speed value (a and b), the secondo with a three seconds:

Note: the motor, without any load, start to turn with 15% of duty cycle.
(As you see from this test, it's needed some time to get a correct speed value)

First test:
a) DC Value:1 Speed value:0
b) DC Value:1 Speed value:0
a) DC Value:2 Speed value:0
b) DC Value:2 Speed value:0
a) DC Value:3 Speed value:0
b) DC Value:3 Speed value:0
a) DC Value:4 Speed value:0
b) DC Value:4 Speed value:0
a) DC Value:5 Speed value:0
b) DC Value:5 Speed value:0
a) DC Value:6 Speed value:0
b) DC Value:6 Speed value:0
a) DC Value:7 Speed value:0
b) DC Value:7 Speed value:0
a) DC Value:8 Speed value:0
b) DC Value:8 Speed value:0
a) DC Value:9 Speed value:0
b) DC Value:9 Speed value:0
a) DC Value:10 Speed value:0
b) DC Value:10 Speed value:0
a) DC Value:11 Speed value:0
b) DC Value:11 Speed value:0
a) DC Value:12 Speed value:0
b) DC Value:12 Speed value:0
a) DC Value:13 Speed value:0
b) DC Value:13 Speed value:0
a) DC Value:14 Speed value:0
b) DC Value:14 Speed value:0
a) DC Value:15 Speed value:0
b) DC Value:15 Speed value:0
a) DC Value:16 Speed value:0
b) DC Value:16 Speed value:0
**a) DC Value:17 Speed value:90
b) DC Value:17 Speed value:150**
a) DC Value:18 Speed value:150
b) DC Value:18 Speed value:165
a) DC Value:19 Speed value:165
b) DC Value:19 Speed value:180
a) DC Value:20 Speed value:180
b) DC Value:20 Speed value:195
a) DC Value:21 Speed value:195
b) DC Value:21 Speed value:210
a) DC Value:22 Speed value:210
b) DC Value:22 Speed value:225
a) DC Value:23 Speed value:225
b) DC Value:23 Speed value:240
a) DC Value:24 Speed value:240
b) DC Value:24 Speed value:240
a) DC Value:25 Speed value:255
b) DC Value:25 Speed value:255
a) DC Value:26 Speed value:255
b) DC Value:26 Speed value:270
a) DC Value:27 Speed value:270
b) DC Value:27 Speed value:285
a) DC Value:28 Speed value:285
b) DC Value:28 Speed value:300
a) DC Value:29 Speed value:300
b) DC Value:29 Speed value:315
a) DC Value:30 Speed value:315
b) DC Value:30 Speed value:315
a) DC Value:31 Speed value:315
b) DC Value:31 Speed value:330
a) DC Value:32 Speed value:330
b) DC Value:32 Speed value:345
a) DC Value:33 Speed value:345
b) DC Value:33 Speed value:360
a) DC Value:34 Speed value:360
b) DC Value:34 Speed value:375
a) DC Value:35 Speed value:375
b) DC Value:35 Speed value:390
a) DC Value:36 Speed value:390
b) DC Value:36 Speed value:405
a) DC Value:37 Speed value:390
b) DC Value:37 Speed value:405
a) DC Value:38 Speed value:405
b) DC Value:38 Speed value:420
a) DC Value:39 Speed value:420
b) DC Value:39 Speed value:435
a) DC Value:40 Speed value:435
b) DC Value:40 Speed value:450
a) DC Value:41 Speed value:450
b) DC Value:41 Speed value:465
a) DC Value:42 Speed value:465
b) DC Value:42 Speed value:480
a) DC Value:43 Speed value:465
b) DC Value:43 Speed value:480
a) DC Value:44 Speed value:480
b) DC Value:44 Speed value:495
a) DC Value:45 Speed value:495
b) DC Value:45 Speed value:510
a) DC Value:46 Speed value:510
b) DC Value:46 Speed value:525
a) DC Value:47 Speed value:525
b) DC Value:47 Speed value:540
a) DC Value:48 Speed value:540
b) DC Value:48 Speed value:540
a) DC Value:49 Speed value:540
b) DC Value:49 Speed value:555
a) DC Value:50 Speed value:555
b) DC Value:50 Speed value:570

Second test (only 20 value)
a) DC Value:1 Speed value:0
b) DC Value:1 Speed value:0
a) DC Value:2 Speed value:0
b) DC Value:2 Speed value:0
a) DC Value:3 Speed value:0
b) DC Value:3 Speed value:0
a) DC Value:4 Speed value:0
b) DC Value:4 Speed value:0
a) DC Value:5 Speed value:0
b) DC Value:5 Speed value:0
a) DC Value:6 Speed value:0
b) DC Value:6 Speed value:0
a) DC Value:7 Speed value:0
b) DC Value:7 Speed value:0
a) DC Value:8 Speed value:0
b) DC Value:8 Speed value:0
a) DC Value:9 Speed value:0
b) DC Value:9 Speed value:0
a) DC Value:10 Speed value:0
b) DC Value:10 Speed value:0
a) DC Value:11 Speed value:0
b) DC Value:11 Speed value:0
a) DC Value:12 Speed value:0
b) DC Value:12 Speed value:0
a) DC Value:13 Speed value:0
b) DC Value:13 Speed value:0
a) DC Value:14 Speed value:0
b) DC Value:14 Speed value:0
a) DC Value:15 Speed value:0
b) DC Value:15 Speed value:0
a) DC Value:16 Speed value:0
**b) DC Value:16 Speed value:135**
a) DC Value:17 Speed value:135
b) DC Value:17 Speed value:150
a) DC Value:18 Speed value:150
b) DC Value:18 Speed value:165
a) DC Value:19 Speed value:165
b) DC Value:19 Speed value:180
a) DC Value:20 Speed value:180

laurensvalk, 2020-07-04:
I’m not sure I see an issue or unexpected result in this data. What is it that concerns you?

It takes some torque to overcome static friction. You should be able to see lower speeds with run() or by moving the motor manually.

We just report the speed value that comes from the motor, and scale it to degrees per second.

GianCann, 2020-07-04:
> What is it that concerns you?

Yes, I also think the result is correct as well.

It is enough to know that a minimum of time is required between the motor start and the speed reading in degrees per second, due to friction.

closed 2020-07-04 by laurensvalk