pybricks/support #85: [Feature] Small user persistent data space

GianCann, 2020-07-03:
There is a way to store very small data between session?
4 or 8 bytes, to save some values to reuse between a execution and another execution.
(For example, the speed value for a motor)

dlech, 2020-07-03:
Although it is technically possible to implement something like this, it is not something we have on the roadmap right now.

GianCann, 2020-07-03:
This is a use case...
Stop the motor at x% speed, and restart (after a shutdown) at the same speed:
https://youtu.be/7qaltulX_iU

laurensvalk, 2020-07-04:
Here’s an alternative which will do what you want in another way:

Just capture the stop button and turn the light off. The hub will look off, and yet it isn’t so you can save as much as you like in memory.

With lights and motors off, the battery should last quite a while. And since your program is persistent, there isn’t much of a loss if the batteries do run out eventually.

So now the feature request is really reduced to saving a tiny bit more battery. You tell us where that should sit on the priority list 😄

GianCann, 2020-07-04:
Hi @laurensvalk ,
I think having a few memory slots to save some bytes can be useful, but you developers are better aware of the priorities and difficulties of implementing new features.

For example, I would prefer to have the UARTDevice class first and the ability to communicate via Bluetooth without interrupting the process with the Input () as it is now.

In any case, as "end user", when I have an idea for a new feature, I tell to you this idea.

After this, are you that evaluate which priority (if any) assign to the idea :smile:

laurensvalk, 2020-07-04:
But what about the alternative solution suggested above? Is there anything that cannot be done that way?

GianCann, 2020-07-04:
> But what about the alternative solution suggested above?

It's a solution, of course.
I can turn off the led, but not the bluetooth radio.
In short, I however waste energy if I leave the hub idle for a long time.

GianCann, 2020-07-04:
> but not the bluetooth radio.

The ability to turn on/off the BLE radio could be another useful feature :smile:

laurensvalk, 2020-07-04:
So it *is* just about power saving then 😄 We’ll probably look at that at some point. There are special low power states that preserve RAM.

I’m teasing a little bit - the main take away is that sometimes a new feature might sound good but often there are existing alternatives that work nearly just as well if you look at the problem from a different angle.

laurensvalk, 2022-10-20:
Among the users that have requested this over [here](https://www.eurobricks.com/forum/index.php?/forums/topic/186132-pybricks-qa/&page=8), ~100 bytes seems to be enough, so we can start with 128 bytes on the Powered Up hubs.

laurensvalk, 2022-10-21:
Proposed API:

```python
# To write:
hub.system.storage(offset=5, data=b"Hello, world!")

# To read:
data = hub.system.storage(offset=5, size=25)
```

We could also add a `storage_view` later on for hubs that can support it, but this should cover the basics.