pybricks/support #87: [Feature] code.pybricks.com enhancement

GianCann, 2020-07-05:
I suggest this enhancement:

- "Clear terminal" button: to delete all the text in the terminal windows.

- "Add to snippet" button: to add a selectes text in the editor to the snippets library

- "Change editor font size" setting: to change the font size of the editor

laurensvalk, 2020-07-05:
To change the font size, zoom in/out using the usual ctrl +/- or ctrl scroll.

Nelvarion, 2020-07-05:
I'd love the ability to copy from the terminal frame.

dlech, 2020-07-05:
> I'd love the ability to copy from the terminal frame.

This works already. Try click-and-drag to select and then right-clicking to get a menu.

Nelvarion, 2020-07-05:
> > I'd love the ability to copy from the terminal frame.
> 
> This works already. Try click-and-drag to select and then right-click to get a menu.

Ctrl+Shift+C it is!

dlech, 2020-07-18:
I split this into one issue per suggestion. #103, #104

closed 2020-07-18 by dlech