pybricks/support #88: [Feature] Input option

JorgePe, 2020-07-05:
Thanks for 'input' - it works great and I can now use it from ev3dev to control hubs.
I know there are better communication options on your roadmap but if there is some time slot I would love a few extra options:

- enable/disable echo
- choose termination characters (CR/LF/CR+LF)

example:
i = input("Prompt", echo = False, end='\r\n')

While echo is useful for debugging, disabling it would reduce a bit the number of messages.
And choosing the proper termination characters would allow apps like Nordic nRF UART or Nordic nRF Toolbox to work out of the box with Pybricks (don't know what is expected by the first but the second only allows those 3 options and none of them seem to work - not even CR that works with my python code, I'm using 0x0D)

laurensvalk, 2020-07-06:
:+1:  for not echo-ing at all. If there isn't an option already I think we can just live without it.

`input` is a standard MicroPython function. As an exercise, do you want to have a look at its code to see what it does and how it might be altered/configured? :smile:

JorgePe, 2020-07-06:
OK. I thought you had implemented it from scratch.

laurensvalk, 2020-07-06:
In general, everything you import from the `pybricks` package is part of this project, and we try to use other existing MicroPython functionality as-is without modification.

closed 2020-07-06 by JorgePe

laurensvalk, 2020-07-06:
@JorgePe since you closed it -- any insights you want to share?

JorgePe, 2020-07-06:
not quite... I didn't find the code itself (supposed to be in builtins module.. where is that?) but documentation only states the prompt option.
I did find input for the unix port but it is a different thing.

laurensvalk, 2020-07-06:
There are quite a few ways to search through MicroPython and it can arguably be a bit daunting at first.

Here's a trick I use sometimes if I can't guess where it might be:

All MicroPython string literals (like function names) are saved only once to save space, prefixed by `MP_QSTR_`. So you can find where `input` is defined by searching for `MP_QSTR_input`. In this particular case, you can see it is guarded by `MICROPY_PY_BUILTINS_INPUT` (just like many optional MicroPython features), which helps you search further. This leads to `mp_builtin_input` in `micropython/py/modbuiltins.c`.

JorgePe, 2020-07-06:
so it call readline and if we want to change the behavior we can implement our own mp_hal_readline... or perhaps just mp_hal_stdin_rx_chr and i'm lost again.
Sorry, I am not a programmer at all :)

laurensvalk, 2020-07-06:
> so it call readline and if we want to change the behavior we can implement our own mp_hal_readline...

Exactly - and that's a good find, and in principle that's even possible within the scope of the project since it does not require modifying MicroPython itself.

What is the current behavior for `input` in `pybricks-micropython` in `ev3dev`? You can run `pybricks-repl` as a shortcut to get the REPL.

> Sorry, I am not a programmer at all :)

I wasn't a programmer when I started this either, I'm probably still not :smile:

JorgePe, 2020-07-06:
> What is the current behavior for input in pybricks-micropython in ev3dev? You can run pybricks-repl as a shortcut to get the REPL. (Honest question - I've never tried myself.)

looks the same - it blocks while waiting for ENTER, accepts just one argument (prompt) or none.

laurensvalk, 2020-07-06:
I meant, does it echo too?

JorgePe, 2020-07-06:
gosh :O
yes, it echoes