pybricks/support #89: [Question] Device ID connected to Hub

GianCann, 2020-07-05:
There is a way to get which device (ID) is connected to a specific port of Hub, instead of catch the exception?

```
#no any motor in port B
try:
   mot=Motor(Port.B)
except Exception as e:
    print("ERRORE: " + str(e))
```

I intend something like
```
if PortA.device== device.ColorDistanceSensor: # or 37
  print("No sensor connected to Port.A")
```

laurensvalk, 2020-07-05:
Yes. Use the `LUMPDevice` class and check the `ID` attribute.

GianCann, 2020-07-05:
Perfect @laurensvalk 
Please add `ID` attribute to the docs of LUMPDevice class ;)

GianCann, 2020-07-05:
Uhmm... It's possible to return -1 in the case of  device non present, instead of throw exception?

laurensvalk, 2020-07-05:
You can't read the `ID` until you've successfully initialized the `LUMPDevice` anyway, so how would that help?

GianCann, 2020-07-05:
I search a 'simple way' to know if there is a device (and which type of this) without using try/catch.

What appened when I use `device=LUMPDevice(Port.A)` ?
It's not possibile to return "-1" if the inizialization is failed?

GianCann, 2020-07-05:
> if there is

Or better, if NOT there is

GianCann, 2020-07-05:
Now, I have to use this code:

```
(imports omitted)
hub=CityHub()

def WarningLED(d0):
    iCount=0
    while iCount < 4:
        hub.light.on(Color.RED)
        wait(d0)
        hub.light.off()
        wait(d0)
        iCount +=1
        

motorPresent=False

try:
   mot=Motor(Port.A)
except Exception as e:
    print("No motor on Port A!")
    try:
        mot=Motor(Port.B)
    except Exception as e:
        print("No motor on Port B!")
    else:
        print("There is a motor on Port B!")
        motorPresent=True
else:
    print("There is a motor on Port A!")
    motorPresent=True
    
if (motorPresent==False):
    WarningLED(100)
else:
    print("Ok, can you run the program!")
```

laurensvalk, 2020-07-05:
> It's not possibile to return "-1" if the inizialization is failed?

This is (Micro)Python, not C. In Python, it is encouraged to be explicit about exceptions. Using try and except to catch *specific* exceptions is perfectly fine. It doesn't make your script "bad", but rather the opposite; it makes it more readable.

If you prefer the C style, you can do something like (I haven't tested this; but you can make something inspired by this):

```python
def is_sensor_connected(port):
    try:
        LUMPDevice(port)
        return True
    except OSError:
        return False

```

GianCann, 2020-07-05:
My goal is to show a warning to the user (with the light on the hub) if there is a device configuration error (no specific motor in a specific port).

I will test with your suggestions

dlech, 2020-07-05:
How about this?

```python
import uerrno

# other imports, etc.

def get_motor(port):
    """Returns a ``Motor`` object if a motor is connected to ``port`` or ``None`` if no motor is connected."""
    try:
        return Motor(port)
    except OSError as ex:
        if ex.args[0] != uerrno.ENODEV:
            raise
        return None

# don't care which port motor is connected to
mot = get_motor(Port.A) or get_motor(Port.B)

if not mot:
    WarningLED(100)
    raise RuntimeError("No motor connected")

print("Ok, can you run the program!")
```

GianCann, 2020-07-05:
> How about this?

> ```python
> # don't care which port motor is connected to
> mot = get_motor(Port.A) or get_motor(Port.B)
> ```

This type of code in new to me! Very interesting!
(This summer I need to learn more deep the Python/MicroPython language... thank also to Pybricks)

Thank you @dlech  & @laurensvalk

laurensvalk, 2020-07-06:
This looks resolved, so closing.

closed 2020-07-06 by laurensvalk