pybricks/support #93: [Question] Strange behavior in cath Exception

GianCann, 2020-07-06:
I run this code

```
(import omitted)
def get_motor(port):
    #Returns a Motor object if a motor is connected to port or None if no motor is connected.
    try:
        return Motor(port)
    except OSError as ex:
        if ex.args[0] != uerrno.ENODEV:
            raise
        return None

#Show a red blinking if not motor present
def warning_LED(speed):
    iCount=0
    while iCount < 4:
        hub.light.on(Color.RED)
        wait(speed)
        hub.light.off()
        wait(speed)
        iCount +=1
        
# don't care which port motor is connected to
mot = get_motor(Port.A) or get_motor(Port.B)

if not mot:
    warning_LED(100)
    raise RuntimeError("No motor connected on Port A or Port B")

crono=StopWatch()
lastClick=0
ramp=0
speedMultipler=2 #with less of 20% of Duty Cycle the motor don't turn
mot.dc(10 * speedMultipler) #start the motor at 20%

while True:
    try:
        if (speedMultipler==10):
            hub.light.on(Color.RED) #with max speed, turn on RED light
        else:
            hub.light.on(Color.GREEN) #otherwise, GREEN light
        wait(200)
    except KeyboardInterrupt as e:
        if (lastClick==0):
            lastClick=crono.time()
        elif ((crono.time() - lastClick) < 250):
            print(str((crono.time() - lastClick)))
            #mot.stop()
            wait(10)
            raise SystemExit("Stop by user..")
        lastClick=crono.time()
        if (ramp==0):    
            speedMultipler +=1
            mot.dc(10 * speedMultipler)
            if (speedMultipler==10):
                ramp=1
        else:
            speedMultipler -=1
            if (speedMultipler>1):
                mot.dc(10 * speedMultipler)
            else:
                mot.stop() 
                raise SystemExit("End of speed ramp cycle...")
```

When I press the green button with a delay > 250 ms between the click, sometime the 'KeyboardInterrupt' exception it's not trap by the try/exception code:
Note:
Line 51 > wait(200)
Line 72 > raise SystemExit("End of speed ramp cycle...")

('pybricks', 'cityhub', '3.0.0a4', '02bee7f on 2020-07-03', 'cityhub')
Traceback (most recent call last):
  File "main.py", line 51, in <module>
KeyboardInterrupt: 
----------
('pybricks', 'cityhub', '3.0.0a4', '02bee7f on 2020-07-03', 'cityhub')
Traceback (most recent call last):
  File "main.py", line 72, in <module>
KeyboardInterrupt: 
----------
('pybricks', 'cityhub', '3.0.0a4', '02bee7f on 2020-07-03', 'cityhub')
Traceback (most recent call last):
  File "main.py", line 72, in <module>
KeyboardInterrupt: 
----------
('pybricks', 'cityhub', '3.0.0a4', '02bee7f on 2020-07-03', 'cityhub')
Traceback (most recent call last):
  File "main.py", line 72, in <module>
KeyboardInterrupt: 
----------
('pybricks', 'cityhub', '3.0.0a4', '02bee7f on 2020-07-03', 'cityhub')
151
Traceback (most recent call last):
  File "main.py", line 59, in <module>
SystemExit: Stop by user..
----------
('pybricks', 'cityhub', '3.0.0a4', '02bee7f on 2020-07-03', 'cityhub')
Traceback (most recent call last):
  File "main.py", line 51, in <module>
KeyboardInterrupt: 

What I miss?

GianCann, 2020-07-06:
I see that the problem goes away by putting a 'wait (500) `on line 51

laurensvalk, 2020-07-06:
Always make sure to post full reproducable examples - not sure why the imports are omitted?

Could you perhaps simplify your code by creating one function of the form ``time_to_press = wait_for_press()``. That way, your code might be easier to read.

GianCann, 2020-07-06:
This is the full code:

```
import uos
from pybricks.hubs import CityHub
from pybricks.pupdevices import Motor
from pybricks.parameters import Port, Stop
from pybricks.parameters import Color
from pybricks.tools import wait, StopWatch
import uerrno

print(uos.uname())

hub=CityHub()

def get_motor(port):
    #Returns a Motor object if a motor is connected to port or None if no motor is connected.
    try:
        return Motor(port)
    except OSError as ex:
        if ex.args[0] != uerrno.ENODEV:
            raise
        return None

#Show a red blinking if not motor present
def warning_LED(speed):
    iCount=0
    while iCount < 4:
        hub.light.on(Color.RED)
        wait(speed)
        hub.light.off()
        wait(speed)
        iCount +=1
        
# don't care which port motor is connected to
mot = get_motor(Port.A) or get_motor(Port.B)

if not mot:
    warning_LED(100)
    raise RuntimeError("No motor connected on Port A or Port B")

crono=StopWatch()
lastClick=lastClick=crono.time()
ramp=0
speedMultipler=2 #with less of 20% of Duty Cycle the motor don't turn
mot.dc(10 * speedMultipler) #start the motor at 20%

while True:
    try:
        if (speedMultipler==10):
            hub.light.on(Color.RED) #with max speed, turn on RED light
        else:
            hub.light.on(Color.GREEN) #otherwise, GREEN light
        wait(500)
    except KeyboardInterrupt as e:
        if ((crono.time() - lastClick) < 300):
            print(str((crono.time() - lastClick))) #debug double click time
            wait(10)
            raise SystemExit("Stop by user..")
        lastClick=crono.time()
        if (ramp==0):    
            speedMultipler +=1
            mot.dc(10 * speedMultipler)
            if (speedMultipler==10):
                ramp=1
        else:
            speedMultipler -=1
            if (speedMultipler>1):
                mot.dc(10 * speedMultipler)
            else:
                raise SystemExit("End of speed ramp cycle...")

```

> Could you perhaps simplify your code by creating...

Ok, I try to do this

laurensvalk, 2020-07-06:
How about editing the original post instead :smile:

GianCann, 2020-07-06:
> by creating one function of the form `time_to_press = wait_for_press()`

I tried to write this function:

```
def isDoubleClick():
    if ((crono.time() - lastClick) < 300):
        print(str((crono.time() - lastClick))) #debug double click time
        return True
    else:
        lastClick=crono.time()
        return False
```

`crono` and `lastClick` are global variables and assigned before call the function.
Why I get this error:

('pybricks', 'cityhub', '3.0.0a4', '02bee7f on 2020-07-03', 'cityhub')
Traceback (most recent call last):
  File "main.py", line 62, in <module>
  File "main.py", line 17, in isDoubleClick
NameError: **local variable referenced before assignment**

GianCann, 2020-07-06:
> NameError: **local variable referenced before assignment**

Ok, I have to use `global` keyword ;)

laurensvalk, 2020-07-07:
Is there still any strange behavior? If so, can you please try to make a shorter/minimal example that reproduces the issue?

GianCann, 2020-07-07:
Ok, @laurensvalk. Today I try to replicate the behavior without the motor.
I can confirm that the behavior is still present with the code shown in the first post.
Randomly, the KeyboardInterrupt exception isn't catched from the Try/Exception code and is raised externally causing the stop of the program.

laurensvalk, 2020-07-07:
Could it be you are pressing the button duing your exception handler? In particular, during the `wait(10)`. You are not catching exceptions there, so any press then will exit the program.

This is why I suggested a dedicated wait for button function, keeping exception handling in one place, so you'll find it easier to see what is going on.

GianCann, 2020-07-07:
The 'anormal' exception is raised in two point, as you see in the previous post:
Line 51:  `wait(200)` (that is internal to Try/exception block)
and
Line 72: ` raise SystemExit("End of speed ramp cycle...")`

So, I can understand when you say "_You are not catching exceptions there_" about the line 72, but I don't understand why this happened on line 51 that IS IN the exception trapping code..

laurensvalk, 2020-07-07:
> So, I can understand when you say "You are not catching exceptions there" about the line 72, but I don't understand why this happened on line 51 that IS IN the exception trapping code..

Thanks, that is indeed interesting. So if you can, try to reproduce in a very minimal example so we can analyse it.

dlech, 2020-07-07:
> but I don't understand why this happened on line 51 that IS IN the exception trapping code..

http://docs.micropython.org/en/latest/genrst/builtin_types.html#exception-in-while-loop-condition-may-have-unexpected-line-number

laurensvalk, 2020-10-07:
Is this resolved? Let us know if we can close it. Thank you.

closed 2021-02-26 by dlech