pybricks/support #95: [Feature] Add a few extra colors to parameters.Color enum

laurensvalk, 2020-07-13:
**Is your feature request related to a problem? Please describe.**
Currently, only a handful of arbitrary colors are available. They are all concentrated in the 0--90 range of the hue spectrum, while the other steps are very course.

The Powered Up color sensors (all kinds) are very capable of discerning more colors.

**Describe the solution you'd like**

Add:
- `Color.MAGENTA`
- `Color.CYAN`
- `Color.GREY`

Keep the default color settings as-is, but make these available in `pybricks.parameters.Color`. That way, by default, the color sensors will still only return primary colors, but the user can add these using the `color_map` if they wish.

`MAGENTA` and `CYAN` are at the same level as `YELLOW`, so it does make sense to have those also. `GREY` is useful to add in brick sorting applications as well.

**Describe alternatives you've considered**
Modify `color_map` to take arbitrary objects instead of `Color` enum members. This might be nice in the long run, but not really worth the effort right now.

**Notes**
We don't really need this at the `pbio` level, so we can defer this until after the light implementation is revised.

laurensvalk, 2020-07-31:
We currently have this for purple, perhaps because it came from legacy pbio:
```c
    PBIO_COLOR_PURPLE = PBIO_COLOR_ENCODE(270, 100, 100), /**< purple */
```

But according to wikipedia, this is violet. Purple is apparently just darker (50%) magenta.

I think we can just keep `270, 100, 100` and call it violet.

-----------------

[image]

[image]

[image]

dlech, 2020-08-05:
Changes released in pybricks-micropython v3.0.0a6

closed 2020-08-05 by dlech