pybricks/support #96: [Bug] Error "`wasm streaming compile failed" running a python program on a movehub / Control+ hub

BertLindeman, 2020-07-13:
**Describe the bug**
Trying to run a stripped sample python program using [https://code.pybricks.com/](https://code.pybricks.com/)

**To reproduce**
1. On Windows 10 open Chrome version 85
2. Flash firmware to the hub (take a minute but seems to end OKE)
3. Click "Connect using bluetooth"
4. The button "Download and run" goes from light blue to dark blue (So is available)
5. Click  button "Download and run"
6. See error:
`wasm streaming compile failed: TypeError: Failed to execute 'compile' on 'WebAssembly': Incorrect response MIME type. Expected 'application/wasm'.,falling back to ArrayBuffer instantiation,Traceback (most recent call last):, File "main.py", line 10, SyntaxError: invalid syntax`

**Expected behavior**
Run the python program 

**Screenshots**
[image]

**Versions**
Google Chrome | 85.0.4183.16 (Official Build) dev (64-bit) (cohort: Dev-nonPGO)
Revision | e5f41a80568bb5f51f3468c3fbb92aac2fceab70-refs/branch-heads/4183@{#226}
Windows 10 OS Version 1909 (Build 18363.900)

`chrome://flags/#enable-experimental-web-platform-features`
 is **enabled**

GianCann, 2020-07-13:
Say hello without the # for comment? ;)

GianCann, 2020-07-13:
You have to use:
#Say hello (this is a comment)
print("Hello, Pybricks!")