pybricks/support #97: [Question]  is the goal to have the same modules with similar functions available for all smart hubs?

BertLindeman, 2020-07-15:
**Question**
Tried on "win10 - Visual Studio code - ev3" to do
`import uos`
`print(uos.uname())`
Like in #53
But the module available on the EV3 does not have uname() available
 
**Context**
Just exploring what can be done with the smart hubs I have.
Would like to have a python program that shows what devices are attached to what port on each hub.
Oke one after another not all at once.

**Screenshots**
[image]

dlech, 2020-07-15:
The EV3 is quite different from the Powered Up hubs, so trying to write code that can run on both would be quite tricky (unless you are just using motors). EV3 has most MicroPython options turned on so it can do quite a bit that Powered Up hubs can't.

Furthermore, EV3 is on Pybricks 2.x and Powered Up hubs are on Pybricks 3.x. This particular case of `uos` is one of the breaking changes between the two. In 2.x `sys.implementation` gives similar information as `uos.uname()` in 3.x.

BertLindeman, 2020-07-15:
Understood @dlech 
Thank you.

Would there already be some debugging program to show the devices connected to the hub?  
Or would it at least possible?

dlech, 2020-07-15:
> Would there already be some debugging program to show the devices connected to the hub?

no

> Or would it at least possible?

Sort of. On Powered Up hubs, you could try to create a [LUMP Device](https://docs.pybricks.com/en/latest/iodevices.html#lump-device) for each port and check the ID. But this only works for UART devices, so you would also have to check `DCMotor` and `Light` to get non-UART devices.

laurensvalk, 2020-07-15:
> But this only works for UART devices

It currently works on *all* devices with an ID, DC Motors included. So maybe we should call it `PUPDevice` instead*, with dc() capability too. Feel free to request this in an issue.

Or, come up with an alternate meaning for U in the acronym L**U**MPDevice so we can keep the name :)

BertLindeman, 2020-07-15:
Thanks guys,

I will take a look at LUMP but it might be out of my league.

In fact you answered my question that there is no intention to synchronize the functions of all hubs.

So closing this question.

closed 2020-07-15 by BertLindeman

laurensvalk, 2020-07-15:
> I will take a look at LUMP but it might be out of my league.

I've been adding example snippets for the sensors lately. I'll make sure the `LUMPDevice` is next on the list.

BertLindeman, 2020-07-15:
@laurensvalk That would be great. Thank you.

I think though that there is already a lot on your plates, seeing the avalange of mails on the pybricks githb ;-)

It's such an exciting project.

Thanks again!

laurensvalk, 2020-07-15:
Our theory is that when we make example snippets, users might contribute example projects for official models one day so we don't have to build them all :)

laurensvalk, 2020-07-15:
@BertLindeman - keep an eye out for the updated docs next week, but you can try this for now:

```python

from pybricks.iodevices import LUMPDevice
from pybricks.pupdevices import ColorDistanceSensor
from pybricks.parameters import Port
from pybricks.tools import wait

def who_am_i(port):
    # Initialize generic device:
    try:
        device = LUMPDevice(port)
    except OSError:
        print("No device on", port)
        return
    
    # Do something depending on what it is:
    if device.ID == 37:
        print("I'm the ColorDistanceSensor and I see: ", end="")
        sensor = ColorDistanceSensor(port)
        print(sensor.color())
    elif device.ID in (48, 49):
        print("I'm a SPIKE motor!")
    else:
        print("I'm another device.")
    
who_am_i(Port.A)
who_am_i(Port.B)

```

BertLindeman, 2020-07-15:
Great Laurens
Thanks!