pybricks/support #99: wait for completion sound

mzajkaniha, 2020-07-16:
by default below command stop program and wait for complete play sound
` ev3.speaker.beep(frequency=800, duration=1000)`

how i can change it to never wait for play sound ?

dlech, 2020-07-18:
Here is one way to do it:

```python
#!/usr/bin/env pybricks-micropython

from _thread import start_new_thread

from pybricks.hubs import EV3Brick
from pybricks.tools import wait

# Create your objects here.
ev3 = EV3Brick()
speaker_lock = allocate_lock()

# Pybricks is not thread-safe so we have to make thread-safe function wrappers
def locked_beep(frequency, duration):
   """Call ev3.speaker.beep with speaker_lock held."""    
    with speaker_lock:
        ev3.speaker.beep(frequency, duration)

def background_beep(frequency, duration):
    """Call ev3.speaker.beep in background thread."""
    start_new_thread(locked_beep, (frequency, duration))

background_beep(300, 1000)

wait(2000)
```

Or with more features:

```python
#!/usr/bin/env pybricks-micropython

from experimental_c import pthread_raise
from _thread import start_new_thread, get_ident, allocate_lock
from usignal import pthread_kill, SIGUSR2

from pybricks.hubs import EV3Brick
from pybricks.tools import wait

# Create your objects here.
ev3 = EV3Brick()
speaker_lock = allocate_lock()

class BackgroundBeep:
    def __init__(self, frequency, duration):
        """Beeps in a background thread.

        Arguments:
            frequency (int): The beep frequency in Hertz.
            duration (int): The duration of the beep in milliseconds.
        """

        self._lock = allocate_lock()
        self._lock.acquire()
        self._thread = start_new_thread(self._beep, (frequency, duration))

    def _beep(self, frequency, duration):
        try:
            with speaker_lock:
                ev3.speaker.beep(frequency, duration)
        finally:
            self._lock.release()

    def wait(self):
        """Waits for the beep to finish."""
        self._lock.acquire()

    def cancel(self):
        """Cancels a beep before it has finished."""
        pthread_raise(self._thread, SystemExit())
        pthread_kill(self._thread, SIGUSR2)

# Write your program here.

# start playing 500Hz beep in background for 5 seconds
beep1 = BackgroundBeep(500, 5000)

# wait for one second
print('begin 1st wait')
wait(1000)
print('end 1st wait')

# try starting another beep in the background - it doesn't start right away
# because the other beep is still busy, so it is "queued" and will play when
# beep1 is finished
beep2 = BackgroundBeep(400, 1000)

# wait for one second again
print('begin 2nd wait')
wait(1000)
print('end 2nd wait')

# stop beep1 - it has only been 2 seconds since it started out of the requested
# 5 seconds
beep1.cancel()

# beep2 starts playing now since beep1 is done

# wait for beep2 to finish
print('begin 3rd wait')
beep2.wait()
print('end 3rd wait')

# total program run time should be 3 seconds
```

l8nite, 2021-12-05:
Hey @dlech - thank you for this example code! I was trying to modify it to play a WAV file (using `ev3.speaker.play_file()` instead of `.beep()` since it seems like there isn't any way to play non-blocking audio files either...

It works decently, but I'm consistently seeing a few errors like `Unhandled exception in thread started by <bound_method ...` or sometimes there's an `ESRCH` signal in response to the `pthread_kill` call. 

Do you know of any more robust way for achieving this? My python multi-threading is pretty weak, and I'm not finding any great documentation on how to achieve what I want with micropython.

dlech, 2021-12-05:
Can you share your full code? Are you using the recommended 22050Hz 16-bit mono wav file format?

l8nite, 2021-12-06:
> Can you share your full code? Are you using the recommended 22050Hz 16-bit mono wav file format?

You know what, as I was cleaning it up to share I found where I introduced a subtle race condition. Once I cleaned it up, the error isn't there anymore and your code is working beautifully :) 

Thanks for the quick reply!

l8nite, 2021-12-06:
If you were curious, the bug was introduced because I was slightly refactoring your class so that it acquired the lock and started the thread in a separate method (instead of on initialization). That way I can play()/stop() to my heart's content on a single instance of the sound class.

l8nite, 2021-12-06:
And just in case anyone sees this in the future and wants the code:

```
from experimental_c import pthread_raise
import _thread
from usignal import pthread_kill, SIGUSR2

speaker_lock = _thread.allocate_lock()

class BackgroundSoundFile:
    def __init__(self, file, loop=False):
        self.file = file
        self.loop = loop
        self.stop_requested = False
        self._lock = _thread.allocate_lock()

    def play(self):
        if not self._lock.locked():
            self._lock.acquire()
            self._thread = _thread.start_new_thread(self._play, (self.file,self.loop))

    def _play(self, file, loop):
        try:
            with speaker_lock:
                ev3.speaker.play_file(file)
                while loop and not self.stop_requested:
                    ev3.speaker.play_file(file)
        finally:
            self._lock.release()

    def wait(self):
        if self._lock.locked():
            self.stop_requested = True # don't loop next time
        self._lock.acquire()

    def stop(self):
        pthread_raise(self._thread, SystemExit())
        pthread_kill(self._thread, SIGUSR2)

```

Usage:

```
sound = BackgroundSoundFile('RAILROAD_CROSSING_BELL_vIF.wav', loop=True)
sound.play()
# do something here while the sound is playing
wait(2000)
sound.stop() # stop the sound
wait(2000) # wait a bit
sound.play() # play the sound again
sound.wait() # wait for the sound to finish
```