pybricks/support #100: pybricks-micropython and evdev

codeadamca, 2020-07-17:
**Question**
Is it possible to use pybricks with evdev? 

**Context**
I'm attempting to use a PS4 controller to control a LEGO creation and I cannot get evdev working with pybricks-micropython. I can get it to work with python 2 and 3. 

Thanks!

laurensvalk, 2020-07-17:
While evdev is not available, it is essentially just reading/writing a few files. You can skip evdev and read the files directly in your script. See e.g. this [demo and code](https://youtu.be/eLGBWOgZFiI).

codeadamca, 2020-07-17:
Awesome, that worked great!

closed 2020-07-17 by codeadamca

laurensvalk, 2020-07-17:
If you're interested, we'd welcome a pull request for a nice and simple example over at https://github.com/pybricks/pybricks-projects/tree/master/snippets/ev3

For example, a program that just reads all the PS4 buttons/sensors in a loop. Then others can easily extend it to their own projects.

For example, it could be somewhat similar to [this snippet](https://github.com/pybricks/pybricks-projects/blob/master/snippets/ev3/bluetooth_read_spike/main.py) [in this project](https://github.com/pybricks/pybricks-projects/tree/master/snippets/ev3/bluetooth_read_spike).

codeadamca, 2020-07-17:
For sure, once I have it all mapped I'll add it.