pybricks/support #101: Can't save from Pybricks Code

ZPhilo, 2020-07-17:
**Describe the bug**
Used to work before, but now when I try to save a program, Chrome Canary pops a window telling me that it blocked the file as it can be dangerous. I can force it to save the file in the download manager, but it's annoying...

**Screenshots**

dlech, 2020-07-17:
This is a web browser security "feature" so not something that we can "fix". However it can be worked around by enabling the "Ask where to save each file before downloading" setting.

[image]

ZPhilo, 2020-07-17:
Thanks David, but...
- "Ask where to save ..." was and is still enabled
- I changed nothing in my configuration (or the folder where I download the files), but "suddendly" I have this problem (even if I load an old file that was saved previously, it fails if I try to save it again).
As said previously it's only annoying since we I can force Chrome to save anyway - or I can use cut/paste in a text editor...

dlech, 2020-07-17:
Bummer, that is annoying.

There is hope for the future though: https://web.dev/native-file-system/

ZPhilo, 2020-07-17:
> 
> There is hope for the future though: https://web.dev/native-file-system/

Ah - another good reason to long for this, my favorite LDraw parts editor will use that in its next version ;)

ZPhilo, 2020-07-21:
Problem seems to have solved itself, probably afer machine reboot (I did restart Chrome, but not my machine when I posted this)

laurensvalk, 2020-10-09:
> Problem seems to have solved itself

This sounds good. We'll close it until it occurs again. Thanks for reporting!

closed 2020-10-09 by laurensvalk