pybricks/support #102: [Bug] ev3-micropython V2  does not auto register EV3-color-sensor

BertLindeman, 2020-07-17:
**Describe the bug**
On ev3-micropython (the system) EV3 and NXT color sensors are not auto-detected.
Hope I am entering this on the correct github repository.
And hope this is not too much info

**To reproduce**
Steps to reproduce the behavior:
1. Running ev3-micropython 2.0
2. on a terminal type the command "dmesg -w"
3. connect the colorsensor to a sensor port, e.g. S3
4. Look at the dmesg output of switching varipus color sensors (last is nxt-color sensor:
`[Sat Jul 18 00:23:13 2020] lego-port port2: Removed device 'ev3-ports:in3:ev3-uart-host'`
`[Sat Jul 18 00:23:28 2020] lego-port port2: Added new device 'ev3-ports:in3:ev3-uart-host'`
`[Sat Jul 18 00:23:50 2020] lego-port port2: Removed device 'ev3-ports:in3:ev3-uart-host'`
`[Sat Jul 18 00:23:55 2020] lego-port port2: Added new device 'ev3-ports:in3:ev3-uart-host'`
`[Sat Jul 18 00:24:00 2020] lego-port port2: Removed device 'ev3-ports:in3:ev3-uart-host'`
`[Sat Jul 18 00:24:03 2020] lego-port port2: Added new device 'ev3-ports:in3:lego-nxt-color'`

**Note missing: "lego-sensor sensor2: Registered 'lego-ev3-color' on 'ev3-ports:in3'"**

5. `ls -l /sys/class/lego-sensor/`
	and see no new sensor has been registered there.

**Expected behavior**
See the color sensor in the /sys/ filesystem AND have the sensor connected in my micropython program

**Screenshots**
not now

**Desktop:**
 - OS and version: [Windows 10 Home 1909]
 - Connection Type: [Wi-Fi]
 - Software: [Visual Studio Code and PuTTY]

**Robot:**
 - Device: EV3
 - Versions: 
Image file:         ev3-micropython-v2.0.0-sd-card-image
Kernel version:     4.14.117-ev3dev-2.3.5-ev3
Brickman:           0.10.3
BogoMIPS:           148.88
Bluetooth:          4.0
Board:              board0
BOARD_INFO_HW_REV=7
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=0016535166E2
BOARD_INFO_TYPE=main

**debugging done so far**
Switched back to ev3dev with the same sensors / motors on the same ports:
`Welcome to ev3dev GNU/Linux 9.12 (stretch) GNU/Linux`
`        Kernel:         4.14.117-ev3dev-2.3.5-ev3 armv5tejl`
`        EV3DEV image:   ev3dev-stretch-ev3-generic-2018-11-13`
and did the same with success.
`dmesg` tail:
`[Fri Jul 17 20:19:53 2020] lego-port port2: Added new device 'ev3-ports:in3:ev3-uart-host'`
`[Fri Jul 17 20:19:56 2020] lego-sensor sensor2: Registered 'lego-ev3-color' on 'ev3-ports:in3'`.

and:
showsensors:
` /sys/class/lego-sensor/sensor2`
` sensor 1 driver_name: lego-ev3-color on port ev3-ports:in3`
`address:              ev3-ports:in3`
`mode:                 COL-REFLECT`
`modes:                COL-REFLECT COL-AMBIENT COL-COLOR REF-RAW RGB-RAW COL-CAL`
`commands:`
`decimals:             0`
`num_values:           1`
`value0:               1`
`units:                pct`
`uevent:`
`LEGO_DRIVER_NAME=lego-ev3-color`
`LEGO_ADDRESS=ev3-ports:in3`

**Background Information**
Trying to program the advanced color sorter of MOC-5615 - Optimized Colour Sorter
The color sensor (both EV3 and NXT) are recognized as seen in dmesg, but not instantiated in the /sys/ filesystem.
And thus not found by micropython.  (My conclusion)

The same **does work** on ev3dev

dlech, 2020-07-17:
The NXT color sensor will never appear in `/sys/class/lego-sensor/`. ev3dev can detect it but there is no driver. It works in Pybricks by using the `raw` mode of the port.

Not sure why the EV3 color sensor is not syncing up though. There really isn't much difference between the two SD card images.

You could just use the working ev3dev SD card. (`sudo apt update && sudo apt install pybricks-micropython` to get the latest Pybricks)

BertLindeman, 2020-07-18:
@dlech Yes, the ev3dev image **does** work, so that would be my circumvention. 
The NXT color sensor was just an attempt to get some more debug info.

If there is anything (easy ;-) ) I can do to debug, let me know.

laurensvalk, 2020-07-18:
Can you try both sensors on different ports?

When you use nxtdevices.ColorSensor, it disables autodetect because it puts the sensor in raw mode. So you will not see anything in the device browser menu after this. Auto id is automatically re-enabled the next time you use ev3devices.ColorSensor.

Other ports are unaffected, so you should still be able to see the EV3 color sensor there.

BertLindeman, 2020-07-18:
> Can you try both sensors on different ports?

@laurens I would forget about the NXT color sensor in this case, unless it os important for debugging. In **this** case I no not need it.

Did a reboot (pybricks SD card) with two EV3 color sensors on S2 and S3 to get a stable situation. 

 `dmesg|grep -i ev3-port`
shows:
> [Sat Jul 18 09:43:59 2020] lego-port port0: Registered 'ev3-ports:in1' on 'ev3-ports:in1'.
[Sat Jul 18 09:43:59 2020] lego-port port1: Registered 'ev3-ports:in2' on 'ev3-ports:in2'.
[Sat Jul 18 09:43:59 2020] lego-port port2: Registered 'ev3-ports:in3' on 'ev3-ports:in3'.
[Sat Jul 18 09:43:59 2020] lego-port port3: Registered 'ev3-ports:in4' on 'ev3-ports:in4'.
[Sat Jul 18 09:43:59 2020] lego-port port4: Registered 'ev3-ports:outA' on 'ev3-ports:outA'.
[Sat Jul 18 09:43:59 2020] lego-port port5: Registered 'ev3-ports:outB' on 'ev3-ports:outB'.
[Sat Jul 18 09:43:59 2020] lego-port port6: Registered 'ev3-ports:outC' on 'ev3-ports:outC'.
[Sat Jul 18 09:43:59 2020] lego-port port7: Registered 'ev3-ports:outD' on 'ev3-ports:outD'.
[Sat Jul 18 09:43:59 2020] lego-port port0: Added new device 'ev3-ports:in1:lego-ev3-touch'
[Sat Jul 18 09:43:59 2020] tacho-motor motor0: Registered 'lego-ev3-l-motor' on 'ev3-ports:outA'.
[Sat Jul 18 09:43:59 2020] lego-port port4: Added new device 'ev3-ports:outA:lego-ev3-l-motor'
[Sat Jul 18 09:43:59 2020] tacho-motor motor1: Registered 'lego-ev3-m-motor' on 'ev3-ports:outD'.
[Sat Jul 18 09:43:59 2020] lego-port port7: Added new device 'ev3-ports:outD:lego-ev3-m-motor'
[Sat Jul 18 09:44:00 2020] lego-port port1: Added new device 'ev3-ports:in2:ev3-uart-host'
[Sat Jul 18 09:44:00 2020] lego-port port2: Added new device 'ev3-ports:in3:ev3-uart-host'
[Sat Jul 18 09:44:46 2020] lego-sensor sensor0: Registered 'lego-ev3-touch' on 'ev3-ports:in1'.

So the touch sensor on S1 is registered and the ev3-color sensors are not.

Tried to look at udev rules, but could not find significant diff between ev3dev and pybricks SD-card images.
But I am no expert on this (far from)

BertLindeman, 2020-07-18:
Maybe a red herring, but I saw this in `journalctl -r`

> Jul 18 10:46:02 ev3dev systemd[1]: Failed to start LEGO MINDSTORMS EV3 UART Sensors.
Jul 18 10:46:02 ev3dev systemd[1]: ev3-uart@ev3x2dports:in2.service: Failed with result 'exit-code'.
Jul 18 10:46:02 ev3dev systemd[1]: ev3-uart@ev3x2dports:in2.service: Control process exited, code=exited, status=1/FAILURE
Jul 18 10:46:02 ev3dev ldattach[1414]: ldattach: cannot open /dev/tty_ev3x2dports:in2: No such file or directory
Jul 18 10:46:02 ev3dev systemd[1]: Starting LEGO MINDSTORMS EV3 UART Sensors...
Jul 18 10:46:02 ev3dev systemd[1]: Unnecessary job for /sys/subsystem/lego/devices/ev3_uart_host_ev3x2dports:in2 was removed.
Jul 18 10:45:59 ev3dev kernel: lego-port port1: Added new device 'ev3-ports:in2:ev3-uart-host'

laurensvalk, 2020-07-18:
Thanks for clarifying that this is only about the EV3 sensors. In that case, yes, it might be an issue with ev3dev. We've been seeing quite a few issues with auto-detect lately, both motors and sensors.

- Does this happen every time? (I've only seen it occasionally and could not consistently reproduce.)
- Does it persist after you unplug the cable and plug it back in?
- Are you certain it happens only on the LEGO EV3 MicroPython version of the image? Both are just ev3dev and with the same kernel.

BertLindeman, 2020-07-18:
@laurensvalk 

> Does this happen every time? (I've only seen it occasionally and could not consistently reproduce.)

Yes constantly

>    Does it persist after you unplug the cable and plug it back in?

Yes

>  Are you certain it happens only on the LEGO EV3 MicroPython version of the image? Both are just ev3dev and with the same kernel. It could be the case, but it could be a coincidence. Retrying a couple times might shed some light on this.

Sure Laurens, ONLY the pybricks image. 

Maybe a hint in my previous comment.

Will re-boot the ev3dev image and re-try a couple of thime to be sure the error is not in that image.

Thanks for your to the point remarks.

laurensvalk, 2020-07-18:
That's very interesting, thanks---Sorry for those questions, I just wanted to rule out a few possibilities we've encountered earlier.

BertLindeman, 2020-07-18:
> Thanks for your to the point remarks.

@laurensvalk I was serious when I wrote this, altough I should have written questions ;-)

Reboot done and each time I dis- an reconnected an ev3-color sensor it got registered OK. (about 10 times each)

Except ONCE when I re-plugged too fast, got numerous:
> serial8250: too much work for irq25

On the ev3dev SD in the journal:
> Jul 18 11:15:58 ev3dev kernel: lego-sensor sensor6: Registered 'lego-ev3-color' on 'ev3-ports:in3'.
Jul 18 11:15:56 ev3dev systemd[1]: Started LEGO MINDSTORMS EV3 UART Sensors.
Jul 18 11:15:56 ev3dev systemd[1]: Starting LEGO MINDSTORMS EV3 UART Sensors...
Jul 18 11:15:55 ev3dev kernel: lego-port port2: Added new device 'ev3-ports:in3:ev3-uart-host'
Jul 18 11:15:47 ev3dev systemd[1]: Stopped LEGO MINDSTORMS EV3 UART Sensors.
Jul 18 11:15:47 ev3dev systemd[1]: Stopping LEGO MINDSTORMS EV3 UART Sensors...
Jul 18 11:15:47 ev3dev kernel: lego-sensor sensor4: Unregistered 'lego-ev3-color' on 'ev3-ports:in3'.
Jul 18 11:15:47 ev3dev kernel: lego-port port2: Removed device 'ev3-ports:in3:ev3-uart-host'

No errors or not registrering seen on this SD card.

dlech, 2020-07-18:
Are both SD cards being tested on the same EV3? Are both SD cards identical? If you flash the opposite image on the SD cards does it make a difference?

BertLindeman, 2020-07-18:
> Are both SD cards being tested on the same EV3?
Yes, just swap the sd-cards on bott. I have only one EV3 ;-) 
No change on cables/sensors at all.

> Are both SD cards identical?

The sd-cards are not the same brand is that what you mean? Or do you mean exactly the same sd-card? I would not know how to determine that, seeing the same firm making the same / different sd-cards under different marketing names and such.
But: I have a Philips 16GB (the ev3dev image) and a Samsung Ultra 16GB (the pybricks image)

I will save / reflash them on a transend UHS-1 16GB SD-card and retest.
Will take some time though as you know.

BertLindeman, 2020-07-18:
Just to be sure I uncoupled the motors.

-  Copied the pybricks image to an SD-card (transend UHS-1 16GB) 
The problem did not go away on the pybrick image.

- Copied the ev3dev image to the **same** SD-card (transend UHS-1 16GB)
 and the ev3dev image did OK as before.

So conclude that the SD-card in this case is irrelevant.

BertLindeman, 2020-07-21:
New debug attempt:

- re-loaded the sd-card image
`ev3micropythonv200sdcardimage-2c980f58ab30715f0659568970baf494.zip`
and flashed.

- step by step installed the packages listed in `apt upgrade` 
Step by step so I could reboot and test in between

- In the end (So update/upgrade completed) the error noted in this issue does **not** surface.

The sd-card that suffers this error has more packages installed then the newly created card.
I have a list of all installed packages on both images, so I could try to sync them.
My first step will be to do an upgrade today (hm not in the last 3 minutes of this day ;-) ) so the "old" card is at the same level.

BertLindeman, 2020-07-22:
Have updated the installed packages of the sd-image with **this** problem as much as possible,
 as far as I think they might be relevant.
This is the remaining diff of the installed packages.
Left the old image with the error, right the newly created image without the error.

Table generated from output of `dpkg --get-selections`

 Old image with error                | .               | . | New image without error                     | .
 --------------------------------------------|---------- | - | -----------------------------------------------|----------
 apt-file                                       | install       | < |  
 apt-show-versions                              | install       | < |  
bsd-mailx                                      | install       | < |
 .       | .    | > | e2fslibs:armel                                |  install
e2fsprogs-l10n                                 | install      | < |
gif2png                                        | install      | < |
libapt-pkg-perl                               |  install       |<|
libbind9-161:armel                          |    install       |<|
libcom-err2:armel                             |  install       |<|
libdns1104:armel                              |  install       |<|
libexporter-tiny-perl                         |  install      | <|
libext2fs2:armel                               | install       |<|
libfstrm0:armel                                | install       |<|
libirs161:armel                                | install      | <
libisc1100:armel                             |   install     |  <
libisccc161:armel                            |   install     |  <
libisccfg163:armel                           |   install     |  <
libjson-c3:armel                              |  install      | <
liblist-moreutils-perl                       |   install     |  <
liblmdb0:armel                               |   install      | <
liblockfile-bin                              |   install      | < |
liblockfile1:armel                          |    install     |  < |
liblwres161:armel                          |     install    |   < |
libnss-systemd:armel                     |       install  |     < |
libprotobuf-c1:armel                      |      install   |    < |
libprotobuf17:armel                        |     install     |  < |
libregexp-assemble-perl                  |       install   |    < |
 .          |  .  | >| linux-image-4.14.111-ev3dev-2.3.3-ev3     |      install
 .           | .  | > | linux-image-4.14.117-ev3dev-2.3.4-ev3    |       install
 .           | .  | > | linux-image-4.14.96-ev3dev-2.3.2-ev3       |     install
 .           | .  | >| mailutils                                      | install
msmtp                                         |  install      | < |
msmtp-mta                                 |      install      | < |
pybricks-micropython                  |          install    |   < |
pybricks-micropython-lib             |           install  |     < |
python3-agt                                  |   install      | < |
python3-protobuf                          |      install      | < | 
. | . |    > | rtl8188eu-modules-4.14.111-ev3dev-2.3.3-ev3  |   install
.                |     .   |                                      > | rtl8188eu-modules-4.14.117-ev3dev-2.3.4-ev3  |   install
.                |     .    |                                     > | rtl8188eu-modules-4.14.96-ev3dev-2.3.2-ev3    |  install
.                |     .     |                                    > | rtl8812au-modules-4.14.111-ev3dev-2.3.3-ev3  |   install
.                |     .      |                                   > | rtl8812au-modules-4.14.117-ev3dev-2.3.4-ev3  |   install
.                |     .       |                                  > | rtl8812au-modules-4.14.96-ev3dev-2.3.2-ev3    |  install
.                |     .        |                                 > | ssmtp                                        |   install

Only `e2fslibs` I cannot get on both images, but I have no idea what to do next to debug this.
Do you have suggestions how to go further? 

Thinking of closing this issue.

dlech, 2020-07-22:
If the bug is not reproducible on a clean install, then we can probably close. It is an ev3dev issue rather than Pybricks issue anyway.

BertLindeman, 2020-07-23:
> If the bug is not reproducible on a clean install, then we can probably close. It is an ev3dev issue rather than Pybricks issue anyway.

@dlech that was the reasoning I could not produce last night. Thank you.

closed 2020-07-23 by BertLindeman

BertLindeman, 2020-07-29:
Not intending to re-open but maybe I am a bit closer to the cause of this....

**scenario**

- ev3 with some sensors

- run my pybricks-python program that addresses the sensorports as `LUMPDevice`

- unplug and re-plug some sensor

The re-plug does not register the sensor automatically.

Looking at the LUMPDevice as a black box, I assume that the LUMPDevice driver sets the mode to `ev3-uart` and thus no autoregister will be done.

My showports before running `debug_LUMP.py`

 port name | intl port dir | port status | driver mode | control path  
 ------------ | :-----------: | -----------: |-------|--------
ev3-ports:in1  | /sys/class/lego-port/port0  |   ev3-uart   |       auto   |       /sys/devices/platform/ev3-ports/ev3-ports:in1/lego-port/port0                 
ev3-ports:in2 |  /sys/class/lego-port/port1  |   ev3-uart  |        auto   |       /sys/devices/platform/ev3-ports/ev3-ports:in2/lego-port/port1                 
ev3-ports:in3 |  /sys/class/lego-port/port2 |    no-sensor      |   auto  |        /sys/devices/platform/ev3-ports/ev3-ports:in3/lego-port/port2                 
ev3-ports:in4  | /sys/class/lego-port/port3  |   ev3-analog   |     auto   |       /sys/class/lego-port/port3/ev3-ports:in4:lego-ev3-touch/lego-sensor/sensor10  
ev3-ports:outA | /sys/class/lego-port/port4  |   tacho-motor  |     auto  |        /sys/class/lego-port/port4/ev3-ports:outA:lego-ev3-m-motor/                 
ev3-ports:outB | /sys/class/lego-port/port5 |    no-motor    |      auto   |       /sys/devices/platform/ev3-ports/ev3-ports:outB/lego-port/port5/               
ev3-ports:outC | /sys/class/lego-port/port6 |    no-motor    |      auto  |        /sys/devices/platform/ev3-ports/ev3-ports:outC/lego-port/port6/               
ev3-ports:outD | /sys/class/lego-port/port7 |    no-motor   |       auto  |        /sys/devices/platform/ev3-ports/ev3-ports:outD/lego-port/port7/               

**After**  running `debug_LUMP.py`

 port name | intl port dir | port status | driver mode | control path  
 ------------ | :-----------: | -----------: |-------|--------
ev3-ports:in1  | /sys/class/lego-port/port0  |   ev3-uart   |       auto  |        /sys/devices/platform/ev3-ports/ev3-ports:in1/lego-port/port0                
ev3-ports:in2  | /sys/class/lego-port/port1   |  ev3-uart     |     auto     |     /sys/devices/platform/ev3-ports/ev3-ports:in2/lego-port/port1                
ev3-ports:in3  | /sys/class/lego-port/port2   |  ev3-uart    |      ev3-uart  |    /sys/devices/platform/ev3-ports/ev3-ports:in3/lego-port/port2                
ev3-ports:in4  | /sys/class/lego-port/port3  |   ev3-uart      |    ev3-uart   |   /sys/devices/platform/ev3-ports/ev3-ports:in4/lego-port/port3                
ev3-ports:outA | /sys/class/lego-port/port4 |    tacho-motor  |     auto  |        /sys/class/lego-port/port4/ev3-ports:outA:lego-ev3-m-motor/                 
ev3-ports:outB | /sys/class/lego-port/port5  |   no-motor     |     auto     |     /sys/devices/platform/ev3-ports/ev3-ports:outB/lego-port/port5/               
ev3-ports:outC | /sys/class/lego-port/port6   |  no-motor    |      auto     |     /sys/devices/platform/ev3-ports/ev3-ports:outC/lego-port/port6/               
ev3-ports:outD | /sys/class/lego-port/port7   |  no-motor   |       auto     |     /sys/devices/platform/ev3-ports/ev3-ports:outD/lego-port/port7/               

On ports in3 and in4 the driver mode changed 

To repair I did:
`echo auto > /sys/devices/platform/ev3-ports/ev3-ports:in4/lego-port/port3/mode`
`echo auto > /sys/devices/platform/ev3-ports/ev3-ports:in3/lego-port/port2/mode`

And the sensors are back again:

port address | sensor path | sensor mode | driver name | driver mode
---| ------------- | ----- | ----------- |-----------
ev3-ports:in2    |    /sys/class/lego-sensor/sensor11   |       COL-REFLECT  |   lego-ev3-color    |        auto
ev3-ports:in4    |    /sys/class/lego-sensor/sensor12   |       TOUCH        |   lego-ev3-touch       |     auto
ev3-ports:in1    |    /sys/class/lego-sensor/sensor7     |      GYRO-ANG   |     lego-ev3-gyro      |       auto

Not sure if the module `LUMPDevice` should set the driver mode back to what it was before?
I know I use `LUMPDevice` in a non conventional way...

dlech, 2020-07-29:
Unfortunately, there isn't a way to guarantee that the port mode is reset to auto when a program exits. This is one of the down sides of using sysfs.

laurensvalk, 2020-07-30:
> Looking at the LUMPDevice as a black box, I assume that the LUMPDevice driver sets the mode to ev3-uart and thus no autoregister will be done.

Correct. For some more background, when you initialize any motor or sensor in Pybricks for ev3dev, this function gets called:

https://github.com/pybricks/pybricks-micropython/blob/master/lib/ev3dev/src/ev3dev_stretch/lego_port.c#L124

Given the ID corresponding to the sensor class that the user chooses, the port gets set appropriately. In case of `LUMPDevice`, see line 181.

Ports are not reset when the program ends. Usually, a user will be using the same sensors on the same sensors across multiple program runs, so resetting would probably not really be needed even if it were reliable and fast. (But it isn't, so when you're frequently restarting your program it is nicer to just stay in the same port mode.)

By the way, the reason we disable auto-detect for `LUMPDevice` is to allow for custom sensors to work that don't have auto ID. Specifically, that makes all of [these projects work on EV3](https://www.steamdiscoverylab.com/gettingstartedwithbackpacks.html).

BertLindeman, 2020-07-30:
Thank you @laurensvalk 
Great explanation.
So now I understand the mechanism behind it, that made me think something was wrong ;-)

Issue was closed correctly.
Bert