pybricks/support #105: Pybricks Code works with MoveHub and a Galaxy Tab

scotus-49, 2020-07-18:
I have Pybricks Code (Beta) working with my **MoveHub** (Lego set 17101), connected by Bluetooth to my Galaxy Tab. I am using "Google Chromium Dev" downloaded from Google Play, and Bluetooth connection works fine. Installing the firmware took less than 5 minutes.

Using my desktop PC running Ubuntu 20.4.1, and a standard install of Chromium (snap) I was also able to test out the motors from my **Lego Boost** set, with a simple code running all 3 simultaneously.

Pybricks Code is an amazing and fantastic tool!

Thank you for all your hard work, and for making this available.

laurensvalk, 2020-07-18:
That's very cool to hear, thanks for sharing!

If you want to help, we welcome pull requests for example programs for all the LEGO sets. So, if you want to make a cool demo of any of the 17101 models that would be pretty cool. Please see #94 for some ideas what we will be looking for.

laurensvalk, 2020-07-18:
Perhaps we need to create a special support template for "Everything is working". It's good to see that from time to time :smile: .

dlech, 2020-07-18:
> standard install of Chromium (snap)

Just curious, did you run into issues with Bluetooth not working with the snap? i.e. https://github.com/pybricks/pybricks-code/issues/125

scotus-49, 2020-07-18:
@laurensvalk 
Yes, a template for "Everything is working" would be nice.

I bought the Logo Boost set last year following an article by [Grumpy-Mike](https://github.com/Grumpy-Mike/Mikes-Pi-Bakery/blob/master/LEGO_Boost/Part%201/mike's_demo.py) in the Raspberry Pi magazine **Magpi** ([Issue 80](https://magpi.raspberrypi.org/issues/80)). Using the [pylgbst](https://github.com/undera/pylgbst) python library from Andrey Pokhilko, I could control Vernie using Bluetooth from the Pi. It refused to work until I ran the python code with admin status, then the sample script had Vernie dancing around and almost running off the table. 

I'm planning at the moment to build a few simple models inspired by Isogawa's **Lego Boost Idea Book**, so that I can try out what works or not with Pybricks. I already use Hugo website creator, and if I can figure out how to use it with Github I would be happy to post some results and code somewhere.

@dlech 
I ran your snap related script on my desktop Linux box, and Chromium initially did not recognize the MoveHub until I after restarted the browser. My Chromium is "Version 84.0.4147.89 (Official Build) snap (64-bit)", and I have just rechecked and the Movehub is seen and useable. Your information in the extra steps for Linux users worked fine for me.

BertLindeman, 2020-07-25:
@scotus-49 a question:
When I switch OFF the movehub and after that want to use the movehub again in pybricks, 
I have to re-download the firmware to the hub before I can program it. 
Before the re-download of the firmware the hub is not recognized via the "Connect" button.... 
In my case either from Windows10 `Google Chrome Version 86.0.4209.2 (Official Build) dev (64-bit)`
or Ubuntu 20.04  `google-chrome-unstable   86.0.4209.2-1 amd64`
Did or do you suffer something similar?

Thanks,
Bert

laurensvalk, 2020-07-25:
@BertLindeman, things recently stopped working for me as well, both 18.04 and 20.04. I have to do some more digging to identify what the issue is, and whether it is the same or not.

BertLindeman, 2020-07-25:
> @BertLindeman, things recently stopped working for me as well, both 18.04 and 20.04. I have to do some more digging to identify what the issue is, and whether it is the same or not.

@laurensvalk Sometimes I saw in the Chrome console that the gatt server was not available. But not each time.
Just now I had to go back to the official LEGO firmware to get connected (on 20.04 this time).
Firmware download in 20.04 did **not** discover the hub. Had to do it on Windows.
Well this is stil **beta** so to be expected from time to time. ;-)
Thanks

scotus-49, 2020-07-26:
@BertLindeman:
I have just rechecked with my Ubuntu (20.04.1 LTS) and am happy to say that everything works as expected. My Movehub was recognised, and and downloaded code started straight away (green LED and motor activation).

When I switch off and then again switch on my MoveHub I have to re-pair with Bluetooth, but the device is recognised and pairing takes place quite fast. I have always pressed the stop button in the browser before switching off the Hub.

I'm using a standard Chromium Version 84.0.4147.89 (Official Build) snap (64-bit), downloaded and installed with the Ubuntu Software app version 3.36.1. I had originally installed and used the Google Chrome Dev browser as recommended, but I am not too keen on using Google software, and thought to try the Chromium version. This has has worked without any problem in the last week, and I happily uninstalled the Google software. Checking in the console just now gave no error messages, only the info about experimental web services.

Checking `bluetoothd -v` in a terminal gives me version 5.53. 

I had found though that using Ubuntu, installing the firmware was **not quite working as expected**. So I used my Galaxy Tab with Google Chrome dev for this (standard install from Google Play), and the download was complete in less than 5 minutes. With my Ubuntu it sort of hung up each time I tried.

BertLindeman, 2020-07-26:
Thank you (how would you like me to address you?), enough to dig through and think about (tomorrow)
 
> I have just rechecked with my Ubuntu (20.04.1 LTS) and am happy to say that everything works as expected. My Movehub was recognised, and and downloaded code started straight away (green LED and motor activation).

Running ` Ubuntu 20.04.1 LTS (GNU/Linux 5.4.0-42-generic x86_64)`

> I'm using a standard Chromium Version 84.0.4147.89 (Official Build) snap (64-bit)

Still using `google-chrome-unstable 86.0.4209.2-1 `

> Checking `bluetoothd -v` in a terminal gives me version 5.53.

Same here.

> I had found though that using Ubuntu, installing the firmware was **not quite working as expected**.

As I wrote in a previous post I had to go to Windows for it.

Well it's beta so we like to find things ;-(  Well I do . . . 

Bert

laurensvalk, 2020-10-09:
This is resolved. Thanks everyone!

> Perhaps we need to create a special support template for "Everything is working". 

This has been added too, so we'll archive this post.

closed 2020-10-09 by laurensvalk