pybricks/support #107: Pybricks Code does not work on the Parpberry Pi

scotus-49, 2020-07-19:
The **Raspberry P**i is at the centre of possibly the world"s largest grouping of coding enthusiasts. So I thought that I would check out whether or not **Pybricks Code** could be made to made to work on this platform. I have several Pi's, and chose a **3B+** as being a widely available model to try with.

I started with my Pi 3B+ and a clean install of a minimized version of the latest OS on a new micro SD Card.

Here is what I found:

I installed the system, and the system ran an auto-update.
The Chromium browser is preinstalled, and I followed the procedure to activate the flags for Experimental Web Platform features.
I opened code.pybricks.com in Chromium.
The connect using bluetooth recognized my MoveHub, but without an orange highlight as per my Linux PC.
Pybricks Code appeared to fail to transmit my code sample, which had been previously checked with my Ubuntu Linux desktop (worked).

I then opened a terminal window, and running "**bluetoothd -v**" in the command line shows version 5.50

Can this be made to work?

dlech, 2020-07-20:
Were there any error messages? Are there any error messages in the console log?

If you need step-by-step instructions on how to view the console log: https://support.google.com/docs/thread/1873663?hl=en

scotus-49, 2020-07-20:
Here's the console log:

Failed to load resource: net::ERR_BLOCKED_BY_CLIENT
readthedocs-doc-embed.js:1 Error loading Read the Docs promo
error @ readthedocs-doc-embed.js:1
2ble-uart.ts:75 Web Bluetooth is experimental on this platform. See https://github.com/WebBluetoothCG/web-bluetooth/blob/gh-pages/implementation-status.md

scotus-49, 2020-07-23:
UPDATE: Partial success - motor running

I have just experimented with a Raspberry Pi 4 and an up-to-date system, running **code.pybricks.com** in the default Chromium browser. After pairing my MoveHub, (it was recognised) and with a simple code running one motor,  the motor started running.

The LED turned green, and the motor turned. However the play and stop buttons on the web-app had greyed out, and stopping the motor was only possible using the Hub button.

I checked the message console, and it was the same as I had previously. But the motor had started this time! 

I know from Mike Cook's article in the Raspberry Pi Magazine Magpi last year (#80) about using Lego Boost with python that he had some problems running the python pylgbst library with the Movehub:

> Here at the Bakery we have two Raspberry Pi boards: one with the current Raspbian release, and the other the previous one. Oddly enough, we found that running normally was fine with the older release, but with the newer one the code needed to be run in supervisor mode in order for it to connect...

I hope to run this through the Raspberry Pi forums to get some  other views . The Pi and Pybricks should be a good match - if this can be made to work.

laurensvalk, 2021-04-08:
Glad to hear it's working! Thanks for getting back. Now we can close this issue.

closed 2021-04-08 by laurensvalk

BertLindeman, 2021-05-15:
Started at raspbian stretch (Debian 9) on a RPI3B+, but the Chromium provided (version 68) in there reported errors on code.pybricks.com.
Did not copy the errors (sorry) but it had to do with `type error keyboard shortcut.replaceall` 
`No function`
After upgrade to `buster` (Debian 10) the version of Chromium went to 90.

Now code.pybricks.com comes up fine.