pybricks/support #108: Pybricks Code Movehub motor.run_angle()

scotus-49, 2020-07-19:
I have made a little experiment with my MoveHub and the Interactive Motor from the Lego Boost set. I have used an idea from Yoshihito Isogawa's book "**The Lego Boost Idea Book**", under "**lifting things*"*, without any changes. I think that posting videos here is probably not possible, so I have a couple of still photos to show the action.  (Admin: how to show videos?)

I have not found the reverse angle action documented anywhere, and in my setup any **clockwise**/**anticlockwise** statements as per the docs produced error messages, but this version worked.

[image]

[image]

I like Isogawa's approach:

"_LEGO bricks aren’t designed to fit in just one specific place, one particular way. Your imagination is your guide when building with LEGO, and you can put bricks and other LEGO pieces together in many ways to build an almost infinite number of creations. After building a model according to the instructions included with your set, try taking the model apart and using its pieces to create a variation of the model—or build a new one altogether. **_That’s where the real world of LEGO begins_**._"

Here is my code (worked out intuitively from the docs):

```
from pybricks.hubs import MoveHub
from pybricks.pupdevices import Motor
from pybricks.parameters import Port, Stop
from pybricks.tools import wait

hub = MoveHub()

# using the interactive motor
motor = Motor(Port.C)
speed = 85

# Say hello :)
print("Starting Pybricks!")

while True:
    motor.run_angle(speed, 95)
 
    # Wait briefly, then reverse
    wait(2000)
    
    motor.run_angle(speed, -95)
    
    # Wait briefly, then reverse
    wait(2000)
```

Any improvements and ideas would be welcome.

dlech, 2020-07-20:
> how to show videos?

As far as I know, GitHub doesn't allow embedding videos. But you could upload it somewhere else and put a link here.

laurensvalk, 2020-07-20:
> Any improvements and ideas would be welcome.

You can play with `run_until_stalled` and `reset_angle` and their parameters to find a fixed initial position. Examples for those will come eventually. We just finished examples for all of the sensors last week.

> clockwise/anticlockwise statements

You can use it like this:

```python
from pybricks.pupdevices import Motor
from pybricks.parameters import Port, Direction
from pybricks.tools import wait

motor = Motor(Port.C, Direction.COUNTERCLOCKWISE)
```

Now, positive speeds/angles will make the motor go counterclockwise instead of the default clockwise.

scotus-49, 2020-07-20:
@dlech 
For videos - I'll look into using Vimeo, if I can work out how to insert a link (or embed)  here. 

@laurensvalk 
Thanks very much for the info - I had tried **Direction** but without including it in the import statement, and of course I got error messages. 

I think I can now try coding an initial motor run to ground the mini platform, and need to look into braking the motor in a raised position.

I must say again that the speed of response using Pybricks Code is fantastic. It allows someone like me to very quickly try out a range of coding solutions, and the almost instantaneous response of the MoveHub lets me see whether I've got it right or not.

laurensvalk, 2020-07-21:
Sounds good!

I just added quite a few examples for the motors. In your case, check example 5 & 6 under the Action category of the motor methods. Curious to hear how that goes for your mechanism.

We'll need to make the examples a little easier to find and navigate, but one thing at a time.

scotus-49, 2020-07-21:
I just found your motor snippets about a half an hour ago. Wonderful! Thank you so much.

I am now starting to catalogue the various properties of the MoveHub and peripherals, so I can expand out my little experiment to use more of the potential available. What I like about the LEGO Boost Creative Toolbox is the mix of standard parts and Technik elements, and now Pybricks Code gives freedom to use some fantasy and imagination.

laurensvalk, 2020-10-07:
This looks to be resolved, so I'll close this issue. Thank you!

closed 2020-10-07 by laurensvalk