pybricks/support #109: Robot class for Walkers

scotus-49, 2020-07-27:
Does Pybricks have a Robotics Class for walking robots? 

I am prototyping a 6-leg simple walker built on the Movehub, using the two built-in motors to drive right and left leg systems independently. I can only find robotics information for wheeled robots in the Pybricks docs. Can this be adapted for walking Robots, with some way of synchronizing the initial motor setup?

Pybricks works great, and my prototype clambers around when I press **run/send**. I'd love to have a bit more variety than just going in a staight(ish) line though, but could use some help.

[image]

All hints would be very much appreciated.

laurensvalk, 2020-07-28:
The plan is that several more generic classes will be added to the `robotics` module.

I can't see from the picture, but does the mechanism work by just running the motors constantly? If so, since your creation uses two motors, you might be able to just use the `DriveBase`. With clever choices of `diameter` and `axle_track`, you can essentialy scale what the motors will do when you use the `straight` and `turn` methods for given values.

These two methods will keep the motors in sync as well.

Does this mechanism provide a way of detecting the right initial configuration? If so, you can make your robot attain that pose before you initialize the `DriveBase`, simply using the invididual motor methods, one leg at a time.

scotus-49, 2020-07-28:
@laurensvalk:
Thank you so much for your advice - helpful as always. Meanwhile my project has evolved a bit further, and now has the interactive motor turning the color/distance sensor (mounted on a vertical rotating axle at the front) to enable the robot to scan its immediate environment in a roughly 120 degree horizontal angle. 

[image]

I've added a few deco elements to take away from the bare skeleton look, and have solved quite a few problems interfacing Lego Technic and the traditional stuff.  But the Boost set contains quite a few useful in-between pieces.

I will definitely look into using the `DriveBase`. I also want to use the fact that the walk mechanism creates pitch and yaw on the MoveHub, and hope to use the built-in movement sensor to vary the leg mechanism motor speeds to give more torque when the higher (orange) centre leg is operational. 

I have to say that the joy in pushing the start/load button and having one's own creation almost instantly moving and reacting to the software coding is quite something. Thanks again for Pybricks!

scotus-49, 2020-07-29:
@laurensvalk:
I've had some partial success with using `DriveBase`, but not the movement I expected. 

Using `straight()` turned my robot round in a circle, while `turn()` marched straight ahead. I swapped over the left and right motors, but the robot did not change direction - clockwise viewed from the top. I experimented with a few different values for the `wheel_diameter`, but nothing changed.

Here's my code:
```
from pybricks.hubs import MoveHub
from pybricks.pupdevices import Motor
from pybricks.parameters import Port, Stop
from pybricks.tools import wait
from pybricks.robotics import DriveBase

hub = MoveHub()

# built-in motors
motor_a = Motor(Port.A)
motor_b = Motor(Port.B)
# interactive motor
motor_c = Motor(Port.C)

print("Experimenting with DriveBase")

myDriveBase = DriveBase(motor_a, motor_b, 10, 100)

# goes forward
myDriveBase.turn(180)

# turns in a circle but not 90 deg.
myDriveBase.straight(90)
```
Have I done something wrong?  

The turning movement (coded with straight) is just what I wanted, and the robot turned nicely around its centre. It was enjoyable to watch.

Francis Sim

laurensvalk, 2020-07-29:
I think you’ll want to set the positive direction of one of the motors as counterclockwise (using the Direction.COUNTERCLOCKWISE when you initialize motor A or B, whichever fits your application best.)

scotus-49, 2020-07-29:
@laurensvalk:
Thanks for the advice - it worked straight away! 

I added `motor_a = Motor(Port.A,  Direction.COUNTERCLOCKWISE)` and my robot marched forward, but only with a negative number in  `myDriveBase.straight()`. A positive number worked as expected - reversing off of my worktop. Other directional combinations for my two motors produced only a turning movement. 

The `turn()` now also works as expected. but I could not get `drive()` to do anything,  possibly it's a coding mistake by me. I have successfully used duty cycle to rotate my front-mounted sensor alternatively left and right.

Now I can start to fine tune-my code. I'm new to multi-legged things, so this is both exploration and fun. Thanks again for **Pybricks**!

laurensvalk, 2020-07-29:
> I added motor_a = Motor(Port.A, Direction.COUNTERCLOCKWISE) and my robot marched forward, but only with a negative number in myDriveBase.straight()

In that case, do it for B, not A.

> I could not get drive() to do anything

If `turn` and `straight` work, `drive` should work also. Keep in mind that `drive` just begins driving and does not wait until a particular target (a straight distance or a turn angle) is complete.

So if `drive` is the last thing in your program, your program ends and the program stops. If this was the case, just add a `wait` with a desired duration of the motion.

scotus-49, 2020-07-29:
Thanks for the information - I'll check this out. 

Is it the case with the MoveHub that when I am running **DriveBase** the third (interactive) motor will not run at the same time? It seems I have an either/or situation with my robot - it's not a problem, just good to know for coding.

laurensvalk, 2020-07-29:
They can be used at the same time. Depending on what you’d like to achieve, perhaps the ‘wait’ argument in some of the motor commands might help.

scotus-49, 2020-07-30:
I still cannot get my **MoveHub** to recognise or do anything with `drive()` from **Pybricks** Robotics class.

Using a bare minimum build, running in front wheel drive mode:

[image]
[image]

and the following code:

```
from pybricks.hubs import MoveHub
from pybricks.pupdevices import Motor
from pybricks.parameters import Port, Stop, Direction
from pybricks.tools import wait
from pybricks.robotics import DriveBase

hub = MoveHub()

# built-in motors
motor_a = Motor(Port.A,  Direction.COUNTERCLOCKWISE)
motor_b = Motor(Port.B)

print("Using the DriveBase:")

myDriveBase = DriveBase(motor_a, motor_b, 29, 65)
myDriveBase.settings(80, 100, 50, 200)

# driving forward
myDriveBase.drive(150, 5)

dist = myDriveBase.distance()
print("Distance traveled is %s mm" %(dist))
```

The console printout reads `Distance traveled is 0 mm` which is quite correct in this case. The LED briefly flashed green, and quickly returned to blue.

For `myDriveBase.straight(150)` everything works nicely, and the programmed distance of 150 measured 150mm after moving.  `distance()` gave me a readout of 151 in the forward direction and 150 in reverse. This is very impressive, since the **Lego Boost** set is essentially sold as a children's toy. Pybricks certainly has a lot under the bonnet!

Is it my coding that's preventing `drive()` from working, or is it something about the MoveHub, or maybe some bugs somewhere? Any help would be appreciated.

scotus-49, 2020-07-30:
I've added the `wait(2000)` and the hub moved forward, but it stopped after about 2 seconds. As expected, a `wait(5000)` makes the hub run for approx. 5 seconds then stops. How do I code continuous movement?

laurensvalk, 2020-07-30:
> How do I code continuous movement?

As in, move forever?

Here is one way to wait until the end of times, or until the batteries run out, whichever comes first:

```python

while True:
    wait(1000)
    # You can do other things in the loop too, like occasionally printing the distance value
```

laurensvalk, 2020-07-30:
You can use the same technique to drive until the sensor detects something:

```python

myDriveBase.drive(150, 5)
while my_sensor.distance() <= 50:
    wait(10)
myDriveBase.stop()
print("I drove for", myDriveBase.distance(), "mm")

```

scotus-49, 2020-07-30:
Thanks - I'd just worked it out myself, and then your reply came in . Sorry, I should have seen that.

Thanks again for your fantastic help!

closed 2020-12-16 by dlech