pybricks/support #110: [Feature] Make Back button on EV3-Brock usable for program

XilefTech, 2020-07-31:
**Describe the solution you'd like**
I think it would be useful to be able to use the back-button on the ev3 brick in your programs too. I thought of a method that would stop it from exiting the program and give it it's own entry in brick.buttons.pressed()

**Describe alternatives you've considered**
Just an idea for a possible feature.

**Additional context**
I would love to use a discrete back-button for some menus and stuff, I'm trying to do on the display

dlech, 2020-07-31:
This is currently not possible for technical reasons. The way ev3dev works would need to be changed. Some related discussion: https://github.com/ev3dev/ev3dev/issues/794.

XilefTech, 2020-08-02:
Okay, so you wrote there:

> It would be simple enough to add a ev3.interrupt_on_back_button() function to ev3dev-lang-python that users could add to the beginning of their program that handles this. It could send SIGINT just like a keyboard interrupt. Then the basic program template would look like...
together with a code snippet.
If my interpretation is correct, you could abuse that to just call a funktion and go back to your main program like 
`from ev3dev import ev3

if __name__ == "__main__":
    try:
        ev3.interrupt_on_back_button()
        # init other stuff

        while True:
            # do stuff
    except:
        # thing to do that sould happen, when the button was pressed
        # just make something to go back into your program`

would that be correct? or is it just something that I overread that would prevent this?
to be clear: I don't plan on implementing it that way in near future but it would be a possible workaround for me if I finished the other stuff on my program xD

Also I'd love to see these changes done in ev3dev so that this would be possible.

devinbreise, 2021-02-09:
So my FLL kids have this large program that does all of their "runs" and controls the whole thing via a master control program so they don't need to "exit" to Brick Man during their competition.

Occasionally, things go wrong and they want to "stop a run", without exiting their "master control" program (because relaunching takes a _long_ time while the clock is ticking).  I'm looking for a solution I can point them at that doesn't involve littering their code with checks against some global state.  If the "back" button on the EV3 were to raise a catchable exception, that would be a very elegant solution for them.  Their "master control" program could simply _try_ each of the "runs" and catch the exception if it were thrown.  But alas, apparently that is not yet an option.

Any other suggestions?

dlech, 2021-02-09:
There is an experimental `run_parallel()` method in v2. https://github.com/pybricks/pybricks-micropython/blob/v2/bricks/ev3dev/modules/pybricks/experimental.py

So maybe you could do something like this (untested)?

```python
from pybricks.experimental import run_parallel

...

def cancel_button_monitor():
    while not any(ev3.buttons.pressed()):
        wait(100)
    throw RuntimeError("cancelled")

def run_1():
    # mission run 1
    ...

def run_2():
    # mission run 2
    ...

def master_control():
    while True:
        # get input
        ...
        try:
            if selected == 1:
                run_parallel(run_1, cancel_button_monitor)
            elif selected == 2:
                run_parallel(run_2, cancel_button_monitor)
        catch Exception as ex:
            print("error during run:")
            print(ex)
        finally:
            # stop all motors here
            ...
```

If a button is pressed during a run, the `cancel_button_monitor()` function will end the `while` loop and raise an exception. When this happens, `run_paralle()l` will raise a `SystemExit` exception in the `run_x()` function to stop it.

devinbreise, 2021-02-09:
Pybricks never disappoints!  We'll give it a try and report back.  Only concern is that they have some pretty tight loops in a few places (scanning for lines at high speed, line follows, etc.).  Presumably running something like cancel_button_monitor() every 100ms wouldn't impact any of that...

dlech, 2021-04-02:
I forgot, there is a file `/etc/brickrun.conf` in ev3dev. If you add a line `delay=1`, this will delay the back button from killing the program by one second. So with this change, the back button should be usable as `Button.LEFT_UP`.

closed 2021-04-02 by dlech