pybricks/support #113: [Question] Pip installed packages not found

XilefTech, 2020-08-02:
**Question**
What am I doing wrong with installing packages with pip?

**Context**
I tried to install a lib that I wanted to use in my project. Through an ssh terminal I used pip install <libName> to install it. Everything went fine until I actually wanted to use the lib in my program. Every time, I would try to import it, I would just get the error that it wasn't found and there was no such library

Please note tho: I tried this with more than one lib and it didn't work with all of them - all the same issue, so it's not something library-specific.

laurensvalk, 2020-08-03:
Pybricks is based on MicroPython. It is a lightweight implementation of Python 3. This makes it fast to load and run, but it does not have all the same features as real Python 3. This also means that libraries installed with `pip` or `pip3` will only be accessible with the real `python`/`python3`.

There is also [upip](https://docs.micropython.org/en/latest/reference/packages.html#upip-package-manager). There's also some background information on what you can use this for. You can follow the instructions I linked to, but you have to replace the `upip` command with `brickrun -r -- pybricks-upip`. So to list the help `-h` you would do:

```
brickrun -r -- pybricks-upip -h
```

But you may just want to look for the library code and include the package in your project folder. There are a lot of different versions, implementations and installation methods for MicroPython packages. If the code is in your project, it's much easier to debug any issues.

dlech, 2020-08-05:
> Edit: This does not work.

On EV3 MicroPython v2.x, the command is `pybricks-upip` (because ev3dev already had `upip` for a different version of MicroPython).

laurensvalk, 2020-08-06:
Please read the crossed out lines again - that does not work.

The help is displayed, but installing gives index errors:

```
brickrun -r -- pybricks-upip install micropython-pystone_lowmem
Traceback (most recent call last):
  File "upip/upip.py", line 314, in <module>
  File "upip/upip.py", line 307, in main
  File "upip/upip.py", line 204, in install
IndexError: string index out of range

```

dlech, 2020-08-06:
I can't see the `-`s when it is crossed out :smile:

dlech, 2021-07-06:
Is this still a problem?

XilefTech, 2021-07-06:
no

closed 2021-07-06 by XilefTech