pybricks/support #114: [Feature] VSCode extension: Reduce uploading time by only uploading the necessary files

XilefTech, 2020-08-03:
**Is your feature request related to a problem? Please describe.**
In my current project, I'm working with many files (mostly image files) in my program, that I'm not changing at all, but every time, I would upload the project to the brick via the VSCode extension, it would upload every single file again, even if it hasn't been changed to the version, already existing on the brick. This takes unnecessary time to complete.

**Describe the solution you'd like**
I'd love to see at least an option in the VSCode extension to only upload files, that are new or have changed, since the last upload

**Describe alternatives you've considered**
Maybe a file, where you can specify certain files or folders (similar to a .gitignore) that will be outright ignored when uploading to the brick. That would be useful too if you got for example a README in your workspace, that the brick doesn't need to run the program.

dlech, 2020-08-03:
This is out of scope of the Pybricks project.

See [this issue](https://github.com/ev3dev/vscode-ev3dev-browser/issues/9] for a feature request for only sending changed file.

The extension also already has [settings](https://github.com/ev3dev/vscode-ev3dev-browser/wiki/Settings) for including and excluding certain files.

XilefTech, 2020-08-04:
Oh thank you, I didn't see that one. Actually I didn't knew that the extension was only a ev3dev thing that pybricks is using too. 

Actually I havn't got a comlete overview of what belongs to ev3dev stuff and what is actually pybricks own stuff, but I'm sure I'll figure that one out too.

Thanks for all the help!

closed 2020-08-04 by XilefTech