pybricks/support #115: [Question] Using EV3 sensor on PUP Hub

GianCann, 2020-08-19:
I tried to read an EV3 Ultrasonic sensor values with LUMPDevice class but I get:

```
Traceback (most recent call last):
  File "main.py", line 15, in <module>
OSError: [Errno 19] ENODEV: 

A sensor or motor is not connected to the specified port:
--> Check the cables to each motor and sensor.
--> Check the port settings in your script.
--> Check the line in your script that matches
    the line number given in the 'Traceback' above.
```

GianCann, 2020-08-19:
Note: the Spike Prime Ultrasonic sensor works well.

dlech, 2020-08-19:
It used to work - at least with EV3 Gryo sensor.

GianCann, 2020-08-19:
I can confirm that the LUMPDevice works well also with the EV3 Color sensor, but not with the EV3 Ultrasonic sensor.
Here you can find the UART presentation data from the sensor:
https://docs.google.com/spreadsheets/d/1fvs8xxCo_nJxBvKCv8mM_bAbfHo4EMdF1gnez5ICaSE/edit?usp=sharing

GianCann, 2020-08-24:
@dlech  here the file!
[EV3UltrasonicSensor.zip](https://github.com/pybricks/support/files/5116963/EV3UltrasonicSensor.zip)

dlech, 2020-08-24:
What software is needed to view the file?

GianCann, 2020-08-24:
@dlech use this: https://sigrok.org/wiki/PulseView

GianCann, 2020-08-24:
Tell me if you need more test ;)

laurensvalk, 2024-03-11:
We could enable this if/when we have sensor modes.

At the moment this won't work because the device manager doesn't recognize EV3 sensors, so never enters LPF2 UART mode.

Also, when skipping the connection manager and going straight to the UART process, something also doesn't work. It just keeps waiting for the sync byte forever.

laurensvalk, 2024-03-11:
I don't necessarily see this as a requirement for Powered Up, but it's a good test case for https://github.com/pybricks/pybricks-micropython/pull/241:

If EV3 sensors work with modern Pybricks drivers on Powered Up, then they will work on EV3 once we implement the UART drivers witg GPIOs to control the buffers.