pybricks/support #117: [Doc] run_until_stalled fails on EV3 medium motor

Walt-White, 2020-08-25:
EV3 Devices documentation should note that the run_until_stalled command only works for EV3 and NXT Large motors, and fails on EV3 Medium motors.
The following code:
ev3 = EV3Brick()
NXT_Large_motor = Motor(Port.A)
EV3_Medium_motor = Motor(Port.D)

ev3.speaker.beep()
NXT_Large_motor.run_until_stalled(250, then=Stop.HOLD, duty_limit=20)
EV3_Medium_motor.run_until_stalled(250, then=Stop.HOLD, duty_limit=20)

Generates the error:
**The requested operation is not valid in the current state:**
listing the line above with the Medium_motor.

Suggested improvement:
Note the limitation in the documentation, which is otherwise very well done.  Pybricks is a welcome environment for the EV3.

laurensvalk, 2020-08-25:
It should also work with the medium motor. Can you try it with a higher duty limit?

It needs a certain minimum to move at all, and that indeed should be better documented.

Walt-White, 2020-08-26:
Thanks, Laurens.  Good call.  I upped the duty limit to 80 and now it works.  I should have thought of doing that earlier; it just appeared to be a configuration issue and not an operational issue.  

I'm new to the Git world and appreciate the Issues format here, but I'm not sure of the etiquette.  Should I just try to delete this entire issue, since it was "pilot error" or do you leave them up?

laurensvalk, 2020-08-26:
> I'm new to the Git world and appreciate the Issues format here, but I'm not sure of the etiquette. Should I just try to delete this entire issue, since it was "pilot error" or do you leave them up?

We welcome all questions, so there's no need to delete anything. Usually we *close* an issue once the problem is solved or the question is answered, but it will remain [available](https://github.com/pybricks/support/issues?q=is%3Aissue+is%3Aclosed) for anyone who searches for similar questions.

In this case we can close this one because it is similar to #45 .

closed 2020-08-26 by laurensvalk