pybricks/support #118: [Project] Community discord as a place to share knowledge and a place to come together for the community

XilefTech, 2020-08-26:
**Description**
Anyone thought of a Pybricks-Discord yet? 
At least I havn't found one so I'm suggesting it here.

I think a discord server would be a great addition to the project. 
Beginners could ask questions and some other more experienced members of the community could help them so that it becomes easier to get started with pybricks coding, even for python-beginners.
Also you could use it as another place to share some cool but smaller ideas or projects. Maybe anyone else got some ideas too.

laurensvalk, 2020-08-26:
Thanks for the suggestion. For now, we have a chat community at gitter. You're welcome to [join](https://gitter.im/pybricks/community).

You can use your GitHub account to sign in.

dlech, 2020-08-26:
We are also keeping an eye on the announced [GitHub Discussions](https://github.blog/2020-05-06-new-from-satellite-2020-github-codespaces-github-discussions-securing-code-in-private-repositories-and-more/#discussions) feature.

dlech, 2021-02-26:
We are now using [GitHub Discussions](https://github.com/pybricks/support/discussions) for questions and sharing projects.

closed 2021-02-26 by dlech