pybricks/support #119: [Feature] Rename Control+ Hub to Technic Hub

laurensvalk, 2020-08-27:
**Is your feature request related to a problem? Please describe.**

It seems that this hub is now officially called the Technic Hub, so we should probably follow suit. We can still change the name now, as long as Pybricks is in beta.

[image]

**Why didn't we call it Technic Hub in the first place?**

- At the time of release, the Technic Hub did not have an official name. It was just a smart hub in the Control+ sets. It wasn't really clear that it was officially in the Powered Up family. According to the screenshot above, it is now.
- The *SPIKE Prime Hub* is/was officially called the [*Technic Large Hub*](https://le-www-live-s.legocdn.com/sc/media/files/support/spike-prime/techspecs_techniclargehub-fba3b469ecb9eaafbde5f24d34ba090e.pdf). This suggested that there would be multiple Technic hubs of different sizes, so just *Technic Hub* could be confusing in the long run. However, most people seem to call it the SPIKE Prime Hub, so that is not really a problem anymore.

**Describe the solution you'd like**
- At first, change only the user facing documentation and MicroPython QSTR that implements the name.
- So, it becomes `from pybricks.hubs import TechnicHub`
- In the long run, update system files and folders to follow suit.

--------

Any :+1:  or :-1: for this change?

GianCann, 2020-08-27:
I think that the correct name is Technic Hub, but I prefer to call it Control+ Hub.
Take also in mind that LEGO call hte Technic Large Hub as "SpikeHub" in they API.
So, what is the best name for it? Honestly, I don't know...
In my personal opinion, you can call Technic Hub/Control+ Hub in the documentation, leaving CPlusHub at API level.

ZPhilo, 2020-08-27:
...and to add to confusion, Technic Large Hub/SpikeHub is going to be used in 51515 "Mindstorms" set!

dlech, 2020-08-27:
It does seem like, according to the LEGO site, that CONTROL+ is the name of the app rather than the name of the hub or product line.

laurensvalk, 2020-08-27:
> ...and to add to confusion, Technic Large Hub/SpikeHub is going to be used in 51515 "Mindstorms" set!

We were thinking to call it Inventor Hub. What do you think?

And the “yellow one” we will probably call Prime Hub, not SPIKE hub. Why? It seems just like we previously had the NXT brick and EV3 brick (not MINDSTORMS brick/hub).

ZPhilo, 2020-08-27:
> We were thinking to call it Inventor Hub. What do you think?

Looks good, though difference should be low compared to Spike Prime hub, except color

> And the “yellow one” we will probably call Prime Hub, not SPIKE hub. Why? It seems just like we previously had the NXT brick and EV3 brick (not MINDSTORMS brick/hub).

I think you're right. Coding software is indeed named Spike, so there might be other products in that range (SPIKE Second? :D )

laurensvalk, 2020-08-27:
If it’s like MINDSTORMS, that could take another 7 years, but I guess we just want to be prepared 😄

laurensvalk, 2020-10-12:
This is done and all pull requests are merged. Docs updated too. 

Just let us know if you have any problems. Thanks!

closed 2020-10-12 by laurensvalk