pybricks/support #120: [Bug] "ImportError: can't import name PFMotor"

BertLindeman, 2020-08-29:
**Describe the bug**
Trying to run snippets/pup/motor_pf_basics.py
Results in
"ImportError: can't import name PFMotor"

**To reproduce**
Steps to reproduce the behavior:
1. download the code from snippets (or copy / paste from the documentation on the right side of Pybricks code)
2. Connect ColorDistanceSensor to port-B
3. Pair to the gray Control+ hub  
4. Run the program
5. See the error...

**Expected behavior**
To be able to control PFmotor(s) using the ColorDistanceSensor
So glad I found this possibility.

**Screenshots**
none so far

**Versions used**
Windows 10
Google Chrome Version 86.0.4240.8 (Official Build) dev (64-bit)
REPL reported version:
MicroPython v3.0.0a5 on 2020-07-15; LEGO TECHNIC Control+ Hub with STM32L431RC

laurensvalk, 2020-08-29:
I think you might be two firmware versions behind. Can you try updating?

BertLindeman, 2020-08-29:
Will do. 
Why did I not think ....

BertLindeman, 2020-08-29:
Bingo! 
Thank you @laurensvalk for the quick and helpful response.

Would it be helpful (or even possible?) to report the firmware version on "pair"

SO glad this works! 

Closing (user error)

closed 2020-08-29 by BertLindeman