pybricks/support #121: [Question] Pairing with Move Hub works only immediately after re-flashing firmware

BertLindeman, 2020-08-30:
**Question**
Pairing in pybricks with Move Hub works only immediately after re-flashing firmware.
After flashing the firmware the move hub led keeps blinking and then pairing **is** possible.
After the hub has been **off**, the led only flashes (some pink color), but the device does not show up on the "connect bluetooth" panel.
The led stops flashing the moment the button is released.
Is this a problem of **my** move hub?

**Context**
Simply trying to pair my Move Hub after it has been off for some time.

-  Windows 10 (64 bit)

-  Google chrome Version 86.0.4240.8 (Official Build) dev (64-bit)
Any debugging possible?

**Screenshots**
none sofar

The PoweredUp hub does not have this problem. (The gray one from The Liebherr)

No idea if this is relative, so just to be sure I add it here:
`DevTools failed to load SourceMap: Could not load content for https://storage.googleapis.com/workbox-cdn/releases/4.3.1/workbox-sw.js.map: HTTP error: status code 404, `net::ERR_HTTP2_PROTOCOL_ERROR`

`DevTools failed to load SourceMap: Could not load content for https://storage.googleapis.com/workbox-cdn/releases/4.3.1/workbox-core.prod.js.map: HTTP error: status code 404, net::ERR_HTTP2_PROTOCOL_ERROR` 

`DevTools failed to load SourceMap: Could not load content for https://storage.googleapis.com/workbox-cdn/releases/4.3.1/workbox-precaching.prod.js.map: HTTP error: status code 404, net::ERR_HTTP2_PROTOCOL_ERROR`

`DevTools failed to load SourceMap: Could not load content for https://storage.googleapis.com/workbox-cdn/releases/4.3.1/workbox-routing.prod.js.map: HTTP error: status code 404, net::ERR_HTTP2_PROTOCOL_ERROR`

`DevTools failed to load SourceMap: Could not load content for https://code.pybricks.com/static/js/2.e78437f0.chunk.js.map: HTTP error: status code 404, net::ERR_HTTP2_PROTOCOL_ERROR`

`DevTools failed to load SourceMap: Could not load content for https://code.pybricks.com/static/css/main.904acb21.chunk.css.map: HTTP error: status code 404, net::ERR_HTTP2_PROTOCOL_ERROR`

`DevTools failed to load SourceMap: Could not load content for https://code.pybricks.com/static/css/2.c7bcc863.chunk.css.map: HTTP error: status code 404, net::ERR_HTTP2_PROTOCOL_ERROR`

`DevTools failed to load SourceMap: Could not load content for https://code.pybricks.com/static/js/main.b8154d2a.chunk.js.map: HTTP error: status code 404, net::ERR_HTTP2_PROTOCOL_ERROR`

The same error 404 ERR_HTTP2_PROTOCOL_ERROR is reported on opening the docs on the right panel.

BertLindeman, 2020-08-30:
Is there a way to see / show the firmware of the Move Hub? REPL is not supported.....

The batteries in the hub are freshly loaded.

laurensvalk, 2020-08-30:
> Is there a way to see / show the firmware of the Move Hub? REPL is not supported.....
> 
> The batteries in the hub are freshly loaded.

from pybricks import version
print(version)

BertLindeman, 2020-08-30:
> from pybricks import version
print(version)

`('movehub', '3.0.0a7', 'v3.0.0a7 on 2020-08-22')`

Thanks @laurensvalk

laurensvalk, 2020-08-30:
> After the hub has been off, the led only flashes (some pink color)

This seems like it is going back into update mode (which should not happen). Does it show up as `LEGO Bootloader` when you click on the `Flash hub firmware` button?

This does not happen on my Move Hub. I wonder if we are perhaps dealing with different bootloader versions or a different checksum.

laurensvalk, 2020-08-30:
Based on your other questions I think you have used Pybricks on the Move Hub before. Did this not happen with previous firmware versions?

laurensvalk, 2020-08-30:
You can try older versions using the method described under ‘saving a program permanently’ in the powered up quick start page.

BertLindeman, 2020-08-30:
> Based on your other questions I think you have used Pybricks on the Move Hub before. Did this not happen with previous firmware versions?

I **did** use pybricks before on this movehub, but I am not so sure as what / where, as there are many parameters involved and I did not take notes ;-(

> You can try older versions using the method described under ‘saving a program permanently’ in the powered up quick start page.

Will try after diner thank you @laurensvalk .

BertLindeman, 2020-08-30:
Some confusion for me:

```
When you’re happy with your final program, you can save it permanently as described below. 
To change the program, just repeat these steps:

Sign in to GitHub.
Go to our GitHub commits.
Click on the green checkmark next to the latest commit. 
```
Here I am lost 
Tried `Click on the latest commit row on the right most button "<>".`
No `details` nor `Artifacts` as indicated...

```
On the dialog that pops up, click details.

Then click Artifacts on the right hand side of the page.
Download the firmware for your hub. The firmware is a ZIP archive containing the basic firmware 
and one main.py script.
Modify the main.py file as you like.
Drag your modified ZIP file onto the firmware update button of the online editor.
The update now proceeds as usual.
```
In the complete zip of the master branch I do not know what file I need.
Sorry...

laurensvalk, 2020-08-30:
> Click on the green checkmark next to the latest commit. 

Ah, right, part of our builds have been failing lately, so there’s now a red cross instead of a green checkmark, sorry about that 😄 

The move hub firmware is still available though. I can create some additional screenshots with instructions tomorrow if needed. Sorry for the confusion.

BertLindeman, 2020-08-31:
No problem @laurensvalk no hurry either.
Now I understand it wasn't me ;-)

BertLindeman, 2020-09-01:
Have been fiddling  with the  movehub.
I assume the problem is windows related.
Did try firmware versions v3.0.0.a1 through v3.0.0.a7.
Directy after the firmware download, the pair works like a charm.

After the movehub has been switched off, a pair does not succeed (No device shows up)

Removed (from windows) the  paired devices.

Trying to pair the movehub to windows has a strange behaviour:
Pressing the green button on the movehub shows (something like)  "LEGO Bootloader: as devicename.
Release the button and re-press immediately shows "pybrickHub" as name.
Then the hub can be paired to pybricjs.

No idea what / why / how. But as it's bedtime, I am no longer very clear, so I stop here for now.

BertLindeman, 2020-09-02:
Installed firmware on the same movehub (I have only one)  on linux mint (20.04.1)
switched off the hub and paired OK.

So I take it that the problem is in windows 10. Maybe MY windows 10?

GianCann, 2020-09-02:
@BertLindeman I'll do a test on my Win10 later

BertLindeman, 2020-09-02:
> @BertLindeman I'll do a test on my Win10 later

Great, thank you.

BertLindeman, 2020-09-02:
@GianCann Looks like I found the culprit: **me**.

To pair I pressed the green button on the movehub **too long** so the hub tries to switch to firmware download mode, not recognized by the "connect using bluetooth" button on pybricks. (not exactly correctly stated, but I assume you understand)

I wonder: did previous firmware require a longer press on the hub button? 
If **not** then I do not know why I am used to do so. Or maybe the hub lighted up later?

Issue can be closed unless @laurensvalk wants to (have someone) do some doc about the firmware download.
Not really I think.

GianCann, 2020-09-02:
Yes, I can confim: nothing problem about pairing MoveHub (one and more time) with latest firmware on Win10

laurensvalk, 2020-09-02:
> I wonder: did previous firmware require a longer press on the hub button?

Just a single click should be enough.

We have been changing how the lights work, however, so maybe that caused some confusion. That work is still ongoing, and we’ll probably update the getting started instructions when we are ready.

Sounds like this is resolved then. Thanks!

closed 2020-09-02 by laurensvalk

BertLindeman, 2020-09-02:
I like the way the hub light reacts on the current firmware (3.0.0.a7) better than the older way. 
Thank you both.