pybricks/support #123: [Bug] PFmotor on channel 1 stops, while on channel 2 through 4 it runs on until stopped

BertLindeman, 2020-09-03:
**Describe the bug**
Running a PFmotor on RED connector on channel 1 behaves differently from a PFmotor on channel 2...4
Running a PFmotor on BLUE connector on channel 1 does **not** show this problem.

- Running on **RED** **and** channel **1**:
  - The PFmotor runs for a short time

- Running on **BLUE** **and** channel **1**:
  - The PFmotor runs until another command is issued.

**Setup**
- controlPlusHub or a movehub with a color-distance sensor on port-D 

- PF IR receiver with a PFmotor on both connectors (RED / BLUE) on a batterybox

**To reproduce**
1. setup Power Functions IR-receiver with a PFmotor (or led) on the REB connector
2. connect a color-distance sensor to port.D on a movehub or CplusHub
3. Pair and such
4. Notice how the motor does stop after a short time, instead of after 3 seconds.

Then switch to channel 2, 3 or 4 and notice the change in behaviour.
Then set the motor on the blue connector and switch to channel 1 and notice this does not show the problem.  

**Expected behavior**
Would expect not difference in running a PFmotor on channel 1, 2, 3, or 4.
Similar to connecting to RED or BLUE.

**Screenshots**
Two movies:

- running RED on channel 1:

- running RED on channel 2:

**Intended to see the movies I pasted here**
Will try to upload or link  later.....

**test program**
Basic program to change channel / RED or BLUE.

```
# PFmotor program for new issue

from pybricks.pupdevices import ColorDistanceSensor, PFMotor
from pybricks.parameters import Port, Color
from pybricks.tools import wait

# Initialize the sensor.
print("define sensor on port D")
sensor = ColorDistanceSensor(Port.D)

cur_chan = 1
print("Initialize motors on channel", cur_chan)
motorR = PFMotor(sensor, cur_chan, Color.RED)
motorB = PFMotor(sensor, cur_chan, Color.BLUE)

def stopMotors():
    print("Brake and stop both motors")
    motorR.brake()
    motorR.stop()
    motorB.brake()
    motorB.stop()

try:
    print("Rotate RED  at power 100.")
    motorR.dc(100)
    wait(3000)
    #print("Rotate BLUE at power 100.")
    #motorB.dc(100)
    #wait(3000)

    print("Rotate RED  at power 50 and stop.")
    motorR.dc(50)
    wait(3000)
    motorR.stop()
    #print("Rotate BLUE at power 50 and stop.")
    #motorB.dc(50)
    #wait(3000)
    motorB.stop()

    wait(1000)

    print("Rotate RED  the other way at half speed, and then stop.")
    motorR.dc(-50)
    wait(3000)
    motorR.stop()
    #print("Rotate BLUE the other way at half speed, and then stop.")
    #motorB.dc(-50)
    #wait(3000)
    motorB.stop()
except KeyboardInterrupt:  # stop the motors on a break
    stopMotors()

# stop motor if no keyboard interrupt has been used:
stopMotors()

```

laurensvalk, 2020-09-03:
Nice catch, thanks! Can you please add the following to your program?

```python
from pybricks.iodevices import LUMPDevice
debug = LUMPDevice(Port.D)
```

You can leave all of your existing code in place. The `LUMPDevice` and `ColorDistanceSensor` class can live side by side without issues.

And then please add this after each of your motor commands:

```python
print(debug.read(7)[0])
```

This should give one number for each command. Please post the output of your script here.

I think you might have found a case where we are sending the wrong value, and this could help us find it.

BertLindeman, 2020-09-03:
The console output (will not help much I think)
**Channel 1**
```
----------
define sensor on port D
Initialize motors on channel 1
0
0
Rotate RED  at power 100.
0
Rotate RED  at power 50 and stop.
0
0
0
Rotate RED  the other way at half speed, and then stop.
0
0
0
Brake and stop both motors
0
0
0
0
```
**Channel 2 output**
```
----------
define sensor on port D
Initialize motors on channel 2
0
0
Rotate RED  at power 100.
0
Rotate RED  at power 50 and stop.
0
0
0
Rotate RED  the other way at half speed, and then stop.
0
0
0
Brake and stop both motors
0
0
0
0
```

The program I used:
```
# PFmotor program for new issue Including debug

from pybricks.pupdevices import ColorDistanceSensor, PFMotor
from pybricks.parameters import Port, Color
from pybricks.tools import wait

# Initialize the sensor.
print("define sensor on port D")
sensor = ColorDistanceSensor(Port.D)

from pybricks.iodevices import LUMPDevice
debug = LUMPDevice(Port.D)

cur_chan = 1
print("Initialize motors on channel", cur_chan)
motorR = PFMotor(sensor, cur_chan, Color.RED)
print(debug.read(7)[0])
motorB = PFMotor(sensor, cur_chan, Color.BLUE)
print(debug.read(7)[0])

def stopMotors():
    print("Brake and stop both motors")
    motorR.brake()
    print(debug.read(7)[0])
    motorR.stop()
    print(debug.read(7)[0])
    motorB.brake()
    print(debug.read(7)[0])
    motorB.stop()
    print(debug.read(7)[0])

try:
    print("Rotate RED  at power 100.")
    motorR.dc(100)
    print(debug.read(7)[0])
    wait(3000)
    #print("Rotate BLUE at power 100.")
    #motorB.dc(100)
    #wait(3000)

    print("Rotate RED  at power 50 and stop.")
    motorR.dc(50)
    print(debug.read(7)[0])
    wait(3000)
    motorR.stop()
    print(debug.read(7)[0])
    #print("Rotate BLUE at power 50 and stop.")
    #motorB.dc(50)
    #wait(3000)
    motorB.stop()
    print(debug.read(7)[0])

    wait(1000)

    print("Rotate RED  the other way at half speed, and then stop.")
    motorR.dc(-50)
    print(debug.read(7)[0])
    wait(3000)
    motorR.stop()
    print(debug.read(7)[0])
    #print("Rotate BLUE the other way at half speed, and then stop.")
    #motorB.dc(-50)
    #wait(3000)
    motorB.stop()
    print(debug.read(7)[0])
except KeyboardInterrupt:  # stop the motors on a break
    stopMotors()

# stop motor if no keyboard interrupt has been used:
stopMotors()

```

BertLindeman, 2020-09-03:
[Movie using channel 1](https://youtu.be/0JgaAkU_5fk)

[Movie using channel 2](https://youtu.be/PVpnUPz_O9E)

BertLindeman, 2020-09-03:
May helps?
Added other debug values after one command: 
```
    motorR.dc(100)
    print(debug.read(0)[0])
    print(debug.read(1)[0])
    print(debug.read(2)[0])
    print(debug.read(3)[0])
    print(debug.read(4)[0])
    print(debug.read(5)[0])
    print(debug.read(6)[0])
    print(debug.read(7)[0])
    print(debug.read(8)[0])
    print(debug.read(9)[0])
    print(debug.read(10)[0])
```
console:
```
Initialize motors on channel 1
0
0
Rotate RED  at power 100.
-1
10
0
0
40
-1
2
0
-1
67
1197
```

laurensvalk, 2020-09-03:
Sorry, it looks like we can't debug it with those values, my bad.

Thanks for the effort you put into the video videos, that helps!

laurensvalk, 2020-09-03:
> I really enjoy the possibility to work with PF motors and leds using PyBricks!

I'm glad to hear it!

A few other questions, not necessarily relevant, just helpful to eliminate other things:

- Does it still happen if you hold/mount the sensor differently?
- There are three varieties of receivers. Can you identify which one you have?

Quoted from [here](https://www.philohome.com/pfrec/pfrec.htm):

> "Power Functions infrared receiver exist in three version. The initial one, had a firmware bug in Single Pin mode. This bug was corrected in second version, identified by **a short blink on the green LED when power is connected**. Finally, a third version with a more powerful motor driver was introduced in 2012 in Rock Crawler set (9398), [which has a V2 mark on it]."

laurensvalk, 2020-09-03:
I can reproduce it. There's certainly something wrong. :tada:

BertLindeman, 2020-09-03:
That's a pity, but I am happy this is not a USR error by me.
Thanks for the notify.
If I can help. let me know.
I tried to capture the infra red code sent by the colordistance sensor but the Arduino IR sensor that I have, has another frequency (I think)

One latest idea: In my program I set the colorsensor led color (on and off).
Might **that** interfere?

laurensvalk, 2020-09-03:
Some experiments, I tried the following 4 cases.

```python

from pybricks.pupdevices import ColorDistanceSensor, PFMotor
from pybricks.parameters import Port, Color
from pybricks.tools import wait

# Initialize the sensor.
sensor = ColorDistanceSensor(Port.C)

# Initialize a motor on channel 1, on the red output.
motor = PFMotor(sensor, 2, Color.RED)

# Rotate and then stop.
motor.dc(100)
wait(4000)
motor.stop()

```

| Channel | Color | Stays on 4 sec | IR Code On, Off |
|---------|-------|----------|---------|
| 1       | RED   |  ❌       |   71, 64      |
| 1       | BLUE  |  ✔️        |    87, 80     |
| 2       | RED   |  ✔️        | 327, 320         |
| 2       | BLUE  |  ✔️        |   343, 336      |

The full meaning of the code can be obtained from this link, but this is the relevant portion for this example:

[image]

laurensvalk, 2020-09-03:
Interestingly, this does seem to work:

```python

from pybricks.parameters import Port
from pybricks.tools import wait
from pybricks.iodevices import LUMPDevice

sensor = LUMPDevice(Port.C)
def send(code):
    sensor.write(7, (code,))

# Turn on channel 1, red, max speed
send(71)
wait(4000)

# Turn off channel 1, red
send(64)

```

That's good news, because then perhaps the sensor is fine. Can you confirm this, @BertLindeman ?

BertLindeman, 2020-09-03:
Forgot to switch port C/D

Results in:
```
Traceback (most recent call last):
  File "main.py", line 10, in <module>
  File "main.py", line 7, in send
OSError: [Errno 95] EOPNOTSUPP: 

The requested operation is not supported on this device:
--> Check the documentation for device compatibility.
--> Check the line in your script that matches
    the line number given in the 'Traceback' above.
```
And the CplusHub or the web page  "hangs".
Disconnect on the web page does not let the hub blink blue, it is still steady blue.
Ok start over ;-)
Press and hold hub button uptil OFF.
Refresh webpage - pair etc.

YES! works like a charm.
**Even** with the sensor conveniently sitting on top of the IR sensor (like in the movie)

laurensvalk, 2020-09-03:
I just pushed a potential fix. You can try it with the usual steps of updating a custom firmware found on the quickstart guide.

And download the fixed firmware file from [here](https://github.com/pybricks/pybricks-micropython/runs/1067166433).

BertLindeman, 2020-09-03:
Will do.
Thank you for the speedy responses.

BertLindeman, 2020-09-03:
Downloaded CplusHub and flashed:
`('cplushub', '3.0.0a7', '969b353 on 2020-09-03')`
Used some of my test programs on channel 1 and 4 just for differenciation.

Works fine.
Will also "do" the movehub and report the result.

laurensvalk, 2020-09-03:
Fix merged in https://github.com/pybricks/pybricks-micropython/commit/969b353a6bb5174a5ca110421e3cc8ac08cbeb4c. Thanks for reporting!

closed 2020-09-03 by laurensvalk