pybricks/support #124: [Feature] Synchronize popups and webpages so they look similar on windows / linux / mac

BertLindeman, 2020-09-03:
**Is your feature request related to a problem? Please describe.**
A very low-level problem: buttons on windows and Linux are not on the same location. 
Please make those panels/popups and such look like the same

**Describe the solution you'd like**
Set buttons on various platforms on the same relative location

**Describe alternatives you've considered**
None sofar

**Additional context**
Look at the location of the buttons in these screenshots:

Pairing popup on windows:
[image]

On Linux (Mint):
[image]

This is a very low priority issue, maybe take a note on the source where these popups are created to make them similar.

There might be other items with similar "problems"

dlech, 2020-09-03:
These dialog boxes are part of the web browser, so we don't have any control over them. The issue would need to be raised at https://bugs.chromium.org/p/chromium/issues/list

BertLindeman, 2020-09-03:
Ah that figures, how the difference occurs.

Thanks you @dlech

laurensvalk, 2021-07-13:
@BertLindeman, does this inconsistency still exist or can we close it?

BertLindeman, 2021-07-13:
@laurensvalk The inconsistency only occurs in dialog boxes as @dlech explained above.
Our own buttons are okay.

So this minor problem "swapped 'pair' and 'cancel' buttons" is only apparent to users swapping back and forth between Windows and Linux. So like me 😉 

This issue depends on boxes provided by the browser, so out of our hands.

For me, this item can be closed.

laurensvalk, 2021-07-13:
I know it’s a Chromium issue, but I was just hoping you could confirm that the buttons are still placed that way or whether they have fixed it by now. :)

dlech, 2021-07-16:
I suspect that if we were to open an issue with Chromium, we would be told that it is working as intended since different OSes have different design patterns (like which order to put the buttons in).

laurensvalk, 2021-07-19:
Then I think we can close this.

closed 2021-07-19 by laurensvalk