pybricks/support #125: [Doc] Document the color codes of the hub lights

BertLindeman, 2020-09-03:
**Describe the issue**
At the moment my "move hub" light is blinking "amber" "amber" "blue" 
I want to know what that means but cannot find it in the pybricks doc.
Currently, the firmware is: `
'movehub', '3.0.0a7', '969b353 on 2020-09-03'`

**Suggested improvement**
I can not easily indicate where a good location in the doc for this kind of information is.
Maybe in  `https://docs.pybricks.com/en/latest/signaltypes.html`

This might be even different per hub type?

If I know where to find the information and how to format it, I am willing to try to write something for it.

dlech, 2020-09-03:
This is something we are actively working on, which is why it isn't documented yet.

So far, we have tried to (mostly) be the same as LEGO, but I don't think they have documented the blink codes either.

Source code: https://github.com/pybricks/pybricks-micropython/blob/master/lib/pbio/sys/light.c

> At the moment my "move hub" light is blinking "amber" "amber" "blue"

This means low battery.

> This might be even different per hub type?

Our aim is to be consistent across all Powered Up hubs.

BertLindeman, 2020-09-03:
> This is something we are actively working on, which is why it isn't documented yet.

Documenting while actively working on the code is contra-productive.
I really appreciate the development effort. 
And if I can help . . . 

> So far, we have tried to (mostly) be the same as LEGO, but I don't think they have documented the blink codes either.

Hmm, it is documented in [LMS user guide on page 8](https://www.lego.com/cdn/cs/set/assets/bltbef4d6ce0f40363c/LMSUser_Guide_LEGO_MINDSTORMS_EV3_11_Tablet_ENUS.pdf)

> This means low battery.

I noticed as some moments later the hub powered off.

> Our aim is to be consistent across all Powered Up hubs.

A good target ;-)

The blink sequences are more or less important, **but**  if you have seen them once, then you know.
But the first time you want to find what it means.

dlech, 2020-09-03:
> Hmm, it is documented in [LMS user guide on page 8](https://www.lego.com/cdn/cs/set/assets/bltbef4d6ce0f40363c/LMSUser_Guide_LEGO_MINDSTORMS_EV3_11_Tablet_ENUS.pdf)

Thanks. However, that is for EV3. The new Powered Up hubs are a bit different.

I also found some info on the [SPIKE Prime lights](https://education.lego.com/en-us/support/spike-prime/faqs#hub) but it is a bit different from the other Powered Up hubs as well.

BertLindeman, 2020-09-03:
;-) That is the problem of trying to condense different hardware platforms. and **so** fun if it works out.

> I also found some info on the SPIKE Prime lights but it is a bit different from the other Powered Up hubs as well.

The orange code is at least the same.
And the blue is related to a connection.

dlech, 2021-05-12:
Lights are documented at https://pybricks.com/install/using-powered-up-hubs/ now (the URL will probably change as we organize the site).

closed 2021-05-12 by dlech