pybricks/support #126: [Question] How to know what modules are available on the movehub

BertLindeman, 2020-09-05:
**Question**
I want to see what modules are available on the hub that I currently use. 
Sometimes a CPlusHub or MoveHub. 

**Context**
Was trying to run a program on the MoveHub, that I wrote on the CPlusHub and got:
```
Running on hw_type: movehub
Traceback (most recent call last):
  File "main.py", in <module>
NameError: name not defined
```
In this case: the culprit was:
```
help('modules')
```

How to find out what modules are available?
Or should I wait for the documentation for things like that to become available?
Like 
- modules available 
- builtin types available (no floating point as example)

**slightly related**
Debugging a program on the MoveHub is a bit clumsy because the linenumber of the problem line is not avaulable.
I know the memory of the MoveHub is limmited. 
Is there a trick available to help a simple programmer?
It would really help if the error name was printed.
Thinking to import something during debug-phase to point to the error line.
(Would it be a good idea to enter a feature request on this?)

 
Running firmware:
('movehub', '3.0.0a8', 'v3.0.0a8 on 2020-09-04')

**Screenshots**
none needed I think.

dlech, 2020-09-06:
You can find the configuration for the Move hub here: https://github.com/pybricks/pybricks-micropython/blob/master/bricks/movehub/mpconfigport.h

Basically it only has `pybricks` modules.

BertLindeman, 2020-09-06:
I assume the movehub has less memory than the CPlusHub, so it is not so easy to get all functionality you want into the firmware.

Thanks for the pointer.

dlech, 2020-09-06:
Yes. The Move hub only has 128kB of flash memory compared to 256kB on the other hubs.

laurensvalk, 2020-09-08:
> How to find out what modules are available? Or should I wait for the documentation for things like that to become available?

This is certainly something we need to document. But before that, we'll have to settle on what to include in the first place.

Eventually I'd like to have a nice visual overview of hubs vs. features, like this:

|                  | Move Hub | City Hub | Control+ Hub | Prime Hub | EV3 (ev3dev) |
|------------------|----------|----------|--------------|-----------|--------------|
| MicroPython      | ✔️        | ✔️        | ✔️            | ✔️         | ✔️            |
| Pybricks Package | ✔️        | ✔️        | ✔️            | ✔️         | ✔️            |
| REPL             | ❌        | ✔️        | ✔️            | ✔️         | ✔️            |
| Helpful errors     | ❌        | ✔️        | ✔️            | ✔️         | ✔️            |
| Floating point   | ❌        | ✔️        | ✔️            | ✔️         | ✔️            |
| Math module      | ❌        | ✔️        | ✔️            | ✔️         | ✔️            |
| Debian Linux     | ❌        | ❌        | ❌            | ❌         | ✔️            |
| Fast boot     | ✔️        | ✔️        | ✔️            | ✔️         | ❌            |
| User RAM         | ? KB     | ? KB     | ? KB         | ? KB      | ? KB         |
| User storage     | ❌        | ❌        | ❌            | ? MB      | ? MB         |

dlech, 2021-07-06:
Tracked in #383.

closed 2021-07-06 by dlech