pybricks/support #127: [Feature] Change color of hub light on power-off.

BertLindeman, 2020-09-05:
**Is your feature request related to a problem? Please describe.**
Would not call it a problem. maybe a wrinkle.
If a hub (CPlusHub in this case) is connected and there is some situation that makes me want to shutdown the hub, the following happens:
I press the green button of the hub. 
The light already was steady blue. 
I keep pressing, but the color will always stay blue.
After I get tired, I release the button and **then** the led goes off.

**Describe the solution you'd like**
After (what is it?) 3 seconds the led goes off and the hub is powered-off.

**Describe alternatives you've considered**
none

**Additional context**
Situation I had was:
Had flashed the a8 firmware to an CPlusHub but the hub did not switch to connecting mode, so no blue blinking but stady blue.
The webpage did not indicate th Bluetooth connection.

dlech, 2020-09-06:
There was a recent regression where status light does not turn off at shutdown until the button is released. However, if the light is solid blue after reboot and doesn't change to flashing, then I don't think it rebooted properly. I see this occasionally as well. So it is possible that the problem is not in our firmware but rather in the bootloader.

BertLindeman, 2020-09-06:
Thanks @dlech 
That is a situation I did not think off.

dlech, 2020-11-17:
Fix committed (not released yet): https://github.com/pybricks/pybricks-micropython/commit/146be836dc4e27029c44f799098e3b2a98eec71e

dlech, 2020-12-02:
Fix was released in firmware v3.0.0a11.

closed 2020-12-02 by dlech