pybricks/support #128: [Question] How to custom-code input/controller

joosbuijsNL, 2020-09-11:
**Question**
I see on the roadmap that Powered Up remote handset support is expected in v4 or later. Is there any way we can custom-build this (or use alpha code) such that we can use this already in the mean time? If desired I could also help a bit in bringing this to pyBricks but would need a bit of a start to get going.

Ideally pyBricks would also support a smartphone as controller, or the different IR remotes that exist.

**Context**
Using pyBricks actually control and drive cars.

GianCann, 2020-09-11:
> Is there any way we can custom-build this (or use alpha code) such that we can use this already in the mean time?

At this moment, you can connect to Pybricks using the nRF UART Service and send commands to the hub.
You can read the commands with mycropython input() function.

Nothing else, for now...

joosbuijsNL, 2020-09-14:
Note to self/others: https://github.com/virantha/bricknil/blob/master/bricknil/hub.py might be a good lead to start understanding the Bluetooth communication.

Question to the devs/community which I could not find myself during the weekend: which Python bluetooth library(/ies) are used in the firmware? ubluetooth? This would help me understand how to program something that can run on the hub.

laurensvalk, 2020-09-14:
We don't really have this capability yet, and probably not for a while. At the moment, a device can connect *to* a hub, but not yet the other way around.

Here's a demo how you can [make two hubs communicate](https://github.com/pybricks/pybricksdev-demo) with a PC in the middle for now.

> Note to self/others: https://github.com/virantha/bricknil/blob/master/bricknil/hub.py might be a good lead to start understanding the Bluetooth communication.

Pybricks does not use this protocol, at least not for now. That protocol is great for remote controlling the hubs from a smart phone / tablet, but in case of Pybricks the code runs on the hub.

Note that you can still nicely make remote-control-like functionality, by just sending small scripts to the hub that do the same thing as the remote control command would do. You can do that with the demo above as well.

We'd like to implement this [simple protocol](https://docs.pybricks.com/en/latest/messaging.html#server-and-client) in the long run, so it is easy do do non-blocking communication.

>  which Python bluetooth library(/ies) are used in the firmware?

There are none. Except for the SPIKE Prime Hub, none of the hubs has enough space for pure Python libraries.

Everything in the firmware is programmed in C, including the `pybricks` package itself, from which you import motors and sensors.

joosbuijsNL, 2020-09-14:
Hi Laurens,

Again, thanks for the lightning-fast and extensive reply.

This helps me to understand the issues why 'just' adding controllers is not feasible, you're working 100% on-brick, which is limited.

The example you mention helps in understanding how the PC-in-the-middle approach could work, thanks a lot! ([The Brick Automation Project](https://www.facebook.com/The-Brick-Automation-Project-2428195644074254/?hc_ref=ARSo_VM2xXZLMHbNqhGqU6Xc2PoJ3hGGXh8JtTXAg2LaUVPNtzwaxVoRoMoVwTqXtfw&fref=nf&__xts__[0]=68.ARD-RlQ_gEMLudYQ7DrE-V_RaTxihQcmiEi4lmwnRsc5e7opBDpW8Mh4SiyumGoWyjWSlrocxjHJYyaY3dD2p072WC4slsPoYNuXPcMFwYgniKyWL4l-3c23JliyGC4UubaSJ1I9JnYaLtBSbIoFRGBVZ2B0BoECZ9Ljq--zpdZcZVr4gpFJXo0cL7v9omqksKQ9WWNx_b39Qvfw5do9ccxVsakKL-PjQ36ZdSJTFbxeb63SigtqgSZOLic78kEfT4djEL1sYkVRIOTd59zbvb1T8NnDwP3neuHOulRceijbskqKb3uAJiVhHxuvIaGAmxxcq2nWHYANNo5jFkAWxjcbNjEW&__tn__=kC-R) seems similar, but then in a GUI style, instead of through Python).

I was trying to understand and implement the low-level Bluetooth protocol, assuming at least something like uBluetooth would be available, this should be feasible. I also came across your C code, which I did not want to get into (Using C and helper functions to build a Python interface looks cumbersome :) ).

I'm looking forward to pyBricks v4 or v5 which hopefully will include controller support, as I feel this is currently lacking (but I fully understand why!).

Keep up the good work!!!