pybricks/support #129: [Question] How to open REPL to PU controller (CPlusHub f.i.) from VSCode on Windows

joosbuijsNL, 2020-09-11:
**Question**
How can I connect to a PoweredUp hub, like the CPlusHub, from VSCode on Windows 10 to run code on there?

**Context**
I'm trying to use VSCode in a similar fassion as code.pybricks.com. This would allow me better/easier file management, code completion, PEP8 linting, etc.

dlech, 2020-09-11:
We are working on a command line tool for this. You can find it at https://github.com/pybricks/pybricksdev

joosbuijsNL, 2020-09-11:
Thanks so much for the lighting-fast reply!

I'm sure I could figure it out but for now it's distracting me too much from the actual playing with Lego, so I'll stick to the web editor, but I'll certainly keep an eye on the dev sub-project.

Keep up the good work!

laurensvalk, 2020-09-14:
> I'm trying to use VSCode in a similar fassion as code.pybricks.com. This would allow me better/easier file management, code completion, PEP8 linting, etc.

Would it help if we created a demo repository to achieve this, so you can clone it and expand it with your own code?

joosbuijsNL, 2020-09-14:
Yes!

(One of the main things I learned is to be 'lazy' in a good way. I guess this will not only be lazy for me, but also for you guys as it potentially saves this type of questions in the future)

Please let me know if I can help, f.i. by testing!

laurensvalk, 2020-09-14:
Sounds good - I'll do that later this week - remind me if it's taking too long :wink:

laurensvalk, 2020-10-07:
Have a look at installing [pybricksdev](https://github.com/pybricks/pybricksdev).

Once installed, open the folder in VS Code. The `Run BLE` configuration will run the current script on the hub.

This would be a good place to start testing.

If it works just fine, you can add it to your own projects by adding `pybricksdev` as a dependency, and just use the same launch configuration.

joosbuijsNL, 2021-01-09:
Just to report that tonight I managed to get it working. Thanks a lot for the effort!

closed 2021-01-09 by joosbuijsNL