pybricks/support #131: [Question] How to use PPS58-Nx in pybricks

ksj537465, 2020-09-17:
**Question**
now im using VScode 

plz let me know how to make code for use PPS58-Nx by micro-python

or can u give me some sample files how to use this sensor

**Context**
create my new project(for experiment)

**Screenshots**
There is a saying that a picture is worth a 1000 words. Screenshots really help to identify and solve problems.

dlech, 2020-09-17:
There is no ev3dev driver for this sensor, so the only option is to use The [I2CDevice class](https://pybricks.github.io/ev3-micropython/iodevices.html#i2c-device). Be sure to read the note on I2C addresses as that is a common mistake to avoid. The [User's Guide](http://www.mindsensors.com/index.php?controller=attachment&id_attachment=218) for the sensor contains the information needed to know which registers to read.

ksj537465, 2020-09-17:
i have a driver for PPS58 from mindsensor look at the file
 
with this problems i have few questions
 
this is my plan
 
pps58 read pressure - datalog pressure(save) - measured pressure is over 16 , motor run - wait for few sec read pressure - datalog - motor run
 
so i have few questions
 
1. how to datalog pps58  pressure data in ev3dev2  or how to use pps58 and datalog in micro-python

2. how to coding measured pressure is over 16 , motor run

ksj537465, 2020-09-25:
download from ev3dev homepage
 
download linux-drivers-ev3dev-buster and ev3dev-lang-python-ev3dev-strech

ksj537465, 2020-12-28:
[image]

i use anconda3 and move the script to vscode

and like this error pop today whats the prblom?

closed 2020-12-28 by ksj537465

laurensvalk, 2020-12-28:
Make sure not to delete the other files in your project folder.

If in doubt, create a new project and just copy the contents of `main.py` to it.