pybricks/support #132: Attempted to connect from ChromeOS Chrome > 85.0.4164.0

phipsitn, 2020-09-24:
Hello, I've been trying to connect my teams chromebooks to ev3 on Pybricks 2.0 via the code.pybricks.com beta.  I've tried several times and have had no luck on connecting.  Also went to chrome://flags and set all bluetooth items to enabled, still no luck.  What is the procedure to connect an ev3 to the web interface?  Thanks.

dlech, 2020-09-24:
Pybricks for EV3 uses a VS Code extension and is not compatible with code.pybricks.com.

I don't know if it is possible to run VS Code + the EV3 extension on ChromeOS

phipsitn, 2020-09-24:
thanks i'll see if I can find a solution.

dlech, 2020-09-28:
It looks like Microsoft has recently started building VS Code for ChromeOS: https://github.com/microsoft/vscode/issues/1031

phipsitn, 2020-09-28:
I managed to find and test that option.  The biggest problem is lack of USB support on the Chromebook for ev3 to use the plugin for vscode.  I’m going to look at using a machine as a server (Mac or pc) for vscode and see if that will allow uploading.

closed 2020-09-29 by phipsitn