pybricks/support #133: Attempted to connect to code.pybricks.com beta from Mac

phipsitn, 2020-09-24:
I've been attempting to connect to ev3 brick with pybricks 2.0 flashed to sd card from my mac.  Tried first from version >85.0.4164.0 with no luck.  Then upgraded to 86.0.4240.55, still no luck.  Went to chrome:\\flags and enabled all bluetooth items.  Still no luck.  What is the procedure on the brick and in chrome to initiate the connection?

dlech, 2020-09-24:
Please see my other response: https://github.com/pybricks/support/issues/132#issuecomment-698045107

dlech, 2020-09-24:
Pybricks for EV3 uses a VS Code extension and is not compatible with code.pybricks.com.

You can find the extension here: https://marketplace.visualstudio.com/items?itemName=lego-education.ev3-micropython

phipsitn, 2020-09-24:
Thanks, I was not sure if that was the case or not.

phipsitn, 2020-09-24:
I also put in a question about connecting from ChromeOS.  I take it that is a non-starter?

closed 2020-09-29 by phipsitn