pybricks/support #134: DCMotor not working on City Hub

laurensvalk, 2020-09-27:
> City hub, Download/Run test program results in error about not seeing a motor on port A.

> Update:  I should have been more specific.  The motor I'm connecting that fails is the Simple Medium Linear Motor (goofy name, eh?) with LEGO Design ID 45303 and Bricklink part number 21980.  It's the motor that came with the City Hub in the PUP Batmobile.

> Update. There's hope! Using the example for train motors, it works!

_Originally posted by @Walt-White in https://github.com/pybricks/support/issues/44#issuecomment-699559451_

laurensvalk, 2020-09-27:
@Walt-White, would you mind sharing your script that produces the error? Thank you!

dlech, 2020-09-27:
The problem was that `Motor` class was used instead of `DCMotor`.

https://github.com/pybricks/support/issues/44#issuecomment-699662577

closed 2020-09-27 by dlech