pybricks/support #136: Web bluetooth won´t connect after windows 10 update

nuno2500, 2020-09-27:
Hello pybricks team 👍 

After updating Windows 10 today, i can connect to the hub but after that the buttons stay light and can´t send the program to the hub. The chrome icon of web bluetooth connection is nonlonger there. I have the same problem on 2 PCs: One was updated and never work well but the other worked well until windows 10 update today.

Never the less, i can do a firmware update (drag the firmware into the FLASH HUB button.

[image]

Thanks for all the work with PyBricks and thanks for the help.

Best regards,

Nuno

Portugal

laurensvalk, 2020-09-27:
Thanks for your report Nuno!

What happens if you refresh the page (F5) or do a hard refresh (CTRL+F5)?

nuno2500, 2020-09-27:
> Thanks for your report Nuno!
> 
> What happens if you refresh the page (F5) or do a hard refresh (CTRL+F5)?

The same. The hub it works with Android.

s-drahcir, 2020-10-04:
I believe that I to am suffering this issue but I am new to this and never tried it before... :)

I have managed to upgrade to the PyBricks firmware. Upon clicking the BLUETOOTH button in the PyBricks toolbar, I can see "PyBricks Hub- Paired". Selecting and then clicking PAIR eventually leads to "Unexpected error while trying to connect. Check console log and report the error.". I assume the console log is shown in the bottom panel and if so, it's empty.

I cannot connect via the BLUETOOTH button or FLASH HUB button until the City Hub itself is turned off.

Things I've tried: Latest Chrome & Chrome Dev version, incognito mode and running as administrator.

Perhaps useful info...

Windows 10 Home - Verison 2004, OS Build 19041.508
Chrome - Version 85.0.4183.121 (Official Build) (64-bit)
Chrome Dev - Version 87.0.4278.0 (Official Build) dev (64-bit)
Intel Wireless Bluetooth - Version 20.100.7.1

RJBrickBuilds, 2020-10-09:
I have the same issue after Windows 10 update. Here's what comes out.
[image]

dlech, 2020-10-09:
I am able to reproduce the issue. 

Microsoft Edge
Version 85.0.564.70 (Official build) (64-bit)

Windows 10 Pro
Version 2004
Build 19041.508

I got a couple of different errors when trying. The second one we have seen before in #56 and #130.

```
error-log.ts:20 DOMException: Connection failed for unknown reason.
kn @ error-log.ts:20
h.context @ redux-saga-core.esm.js:424
z.<computed> @ redux-saga-core.esm.js:578
(anonymous) @ redux-saga-core.esm.js:1203
m @ redux-saga-core.esm.js:1271
p @ redux-saga-core.esm.js:1161
c @ redux-saga-core.esm.js:1251
l @ redux-saga-core.esm.js:503
e.put @ redux-saga-core.esm.js:339
L.e.put @ redux-saga-core.esm.js:371
(anonymous) @ redux-saga-core.esm.js:1412
dispatch @ redux.js:638
(anonymous) @ io-6de156f3.js:133
(anonymous) @ redux-saga-core.esm.js:472
d @ redux-saga-core.esm.js:31
b @ redux-saga-core.esm.js:87
p @ redux-saga-core.esm.js:46
z.<computed> @ redux-saga-core.esm.js:468
(anonymous) @ redux-saga-core.esm.js:1203
m @ redux-saga-core.esm.js:1271
p @ redux-saga-core.esm.js:1161
c @ redux-saga-core.esm.js:1251
(anonymous) @ redux-saga-core.esm.js:396
```

```
error-log.ts:20 DOMException: GATT Server is disconnected. Cannot retrieve services. (Re)connect first with `device.gatt.connect`.
kn @ error-log.ts:20
h.context @ redux-saga-core.esm.js:424
z.<computed> @ redux-saga-core.esm.js:578
(anonymous) @ redux-saga-core.esm.js:1203
m @ redux-saga-core.esm.js:1271
p @ redux-saga-core.esm.js:1161
c @ redux-saga-core.esm.js:1251
l @ redux-saga-core.esm.js:503
e.put @ redux-saga-core.esm.js:339
L.e.put @ redux-saga-core.esm.js:371
(anonymous) @ redux-saga-core.esm.js:1412
dispatch @ redux.js:638
(anonymous) @ io-6de156f3.js:133
(anonymous) @ redux-saga-core.esm.js:472
d @ redux-saga-core.esm.js:31
b @ redux-saga-core.esm.js:87
p @ redux-saga-core.esm.js:46
z.<computed> @ redux-saga-core.esm.js:468
(anonymous) @ redux-saga-core.esm.js:1203
m @ redux-saga-core.esm.js:1271
p @ redux-saga-core.esm.js:1161
c @ redux-saga-core.esm.js:1251
l @ redux-saga-core.esm.js:503
take @ redux-saga-core.esm.js:183
H.<computed> @ redux-saga-core.esm.js:507
(anonymous) @ redux-saga-core.esm.js:1203
m @ redux-saga-core.esm.js:1271
p @ redux-saga-core.esm.js:1161
c @ redux-saga-core.esm.js:1251
(anonymous) @ redux-saga-core.esm.js:396
Promise.then (async)
T @ redux-saga-core.esm.js:395
z.<computed> @ redux-saga-core.esm.js:527
(anonymous) @ redux-saga-core.esm.js:1203
m @ redux-saga-core.esm.js:1271
p @ redux-saga-core.esm.js:1161
c @ redux-saga-core.esm.js:1251
(anonymous) @ redux-saga-core.esm.js:592
m @ redux-saga-core.esm.js:56
z.<computed> @ redux-saga-core.esm.js:584
(anonymous) @ redux-saga-core.esm.js:1203
m @ redux-saga-core.esm.js:1271
p @ redux-saga-core.esm.js:1161
c @ redux-saga-core.esm.js:1251
Promise.then (async)
T @ redux-saga-core.esm.js:395
z.<computed> @ redux-saga-core.esm.js:527
(anonymous) @ redux-saga-core.esm.js:1203
m @ redux-saga-core.esm.js:1271
p @ redux-saga-core.esm.js:1161
c @ redux-saga-core.esm.js:1251
Promise.then (async)
T @ redux-saga-core.esm.js:395
z.<computed> @ redux-saga-core.esm.js:527
(anonymous) @ redux-saga-core.esm.js:1203
m @ redux-saga-core.esm.js:1271
p @ redux-saga-core.esm.js:1161
U @ redux-saga-core.esm.js:1108
(anonymous) @ redux-saga-core.esm.js:585
m @ redux-saga-core.esm.js:56
z.<computed> @ redux-saga-core.esm.js:584
(anonymous) @ redux-saga-core.esm.js:1203
m @ redux-saga-core.esm.js:1271
p @ redux-saga-core.esm.js:1161
c @ redux-saga-core.esm.js:1251
l @ redux-saga-core.esm.js:503
e.put @ redux-saga-core.esm.js:339
L.e.put @ redux-saga-core.esm.js:371
(anonymous) @ redux-saga-core.esm.js:1412
dispatch @ redux.js:638
(anonymous) @ io-6de156f3.js:133
(anonymous) @ redux-saga-core.esm.js:472
d @ redux-saga-core.esm.js:31
b @ redux-saga-core.esm.js:87
p @ redux-saga-core.esm.js:46
L.e.put @ redux-saga-core.esm.js:375
(anonymous) @ redux-saga-core.esm.js:1412
onAction @ BluetoothButton.tsx:38
onClick @ ActionButton.tsx:34
a @ react-dom.production.min.js:14
d @ react-dom.production.min.js:14
(anonymous) @ react-dom.production.min.js:14
g @ react-dom.production.min.js:15
st @ react-dom.production.min.js:52
it @ react-dom.production.min.js:51
at @ react-dom.production.min.js:52
dt @ react-dom.production.min.js:56
I @ react-dom.production.min.js:287
$ @ react-dom.production.min.js:19
Jt @ react-dom.production.min.js:70
Zt @ react-dom.production.min.js:69
t.unstable_runWithPriority @ scheduler.production.min.js:19
Fi @ react-dom.production.min.js:122
H @ react-dom.production.min.js:287
Yt @ react-dom.production.min.js:68
Show 43 more frames from Library code
```

dlech, 2020-10-09:
Has anyone tracked down the specific update that started the issue?

RJBrickBuilds, 2020-10-09:
I'd love to but I wouldn't have a clue how to do it. However, I removed the updates in my desktop and it is now working well. However, in my Surface Pro, I couldn't do the same. I did reset my Windows for my Surface Pro but ended up with the same issue. I uninstalled all the updates that I can and the issue still persists. However, I can't remove the updates below. Could it be one of these: 
[image]

dlech, 2020-10-10:
Apparently Windows started sending a new BLE request to our firmware that we haven't seen before. So we were able to fix this in the firmware. So this means everyone will need to upgrade their firmware to v3.0.0a9. See https://github.com/pybricks/support/issues/48#issuecomment-706463541 for release notes (important breaking change!).

closed 2020-10-10 by dlech

RJBrickBuilds, 2020-10-10:
Appreciate this a lot! It works! Thanks! :)