pybricks/support #146: [Bug] error: "bad checksum" in hub.ts:100

BertLindeman, 2020-10-10:
**Describe the bug**
Was using the REPL and wanted to run my program again.
As I do not see how to get out of the REPL, 
I just pressed the "Download and run this program"-button
Resulting in error "bad checksum" 

**Environment**
Windows 10
Edge 86
MicroPython v3.0.0a9 on 2020-10-10; LEGO Technic Hub with STM32L431RC

**To reproduce**
After loading firmware connect to the hub, (No devices connected to the hub)
and then:
1. Press "Start REPL in terminal"
2. Type help() and enter
3. Press "Download and run this program"
4. See error and see status light on the hub stay green.

**Expected behavior**
My program to run. 

**Screenshots**
[image]

[image]

[image]

**maybe not related**
The program I had loaded:
```
#!/usr/bin/env pybricks-micropython

from pybricks.tools import wait

from pybricks.iodevices import LUMPDevice
try:
    from pybricks.hubs import EV3Brick
    EV3_available = True
except:
    EV3_available = False
    print("could not import EV3Brick")

try:
    from pybricks.pupdevices import ColorDistanceSensor, Motor, ColorSensor
    from pybricks.pupdevices import UltrasonicSensor
    from pybricks.pupdevices import ForceSensor
except:
    print("could not import pupdevices")

from pybricks.parameters import Port
# from pybricks.tools import wait

# print(LUMPDevice.__dict__)

try:
    ev3 = EV3Brick()
except Exception as e:
    print("No EV3 brick here, due to: ", e, end="; ")
    print("Not imported on THIS hub")

def who_am_i(xport):  # noqa: C901
    #  try to initialize generic device:
    try:
        # print("try Port:", end="")
        # print(xport)
        device = LUMPDevice(xport)
    except OSError as e:
        print("OSError on:", xport, ":",  end="")
        print(e)
        return
    except ValueError as e:
        print("Exception ", end="")
        print(e.__class__, end="")
        print(" occurred on", xport, ": ", end="")
        print(e)
        return
    except Exception as e:
        print("General ", e.__class__, "occurred.")
        print(" on ", xport, ": ", end="")
        print(e)
        return
    else:
        # Do something depending on what it is:
        print("device", device)
        # print("device.ID:", device.ID)
        # dir(device)
        try:
            print("%-7s device.ID=%2d " % (xport, device.ID), end="")
        except ValueError as e:
            print("Exception ", end="")
            print(e.__class__, end="")
            print(" occurred on", xport, ": ", end="")
            print(e)
            return
        except Exception as e:
            print("Exception", end="")
            print(e.__class__, end="")
            print("occurred on", xport, ": ", end="")
            print(e)
            return
        if device.ID == 0:
            print("NO device; OR recycle the hub")
        elif device.ID == 8:
            print("ID 8 should be 'NO device' or recycle the hub")
        elif device.ID == 37:
            print("BOOST Color and Distance Sensor \t\tresponse: ", end="")
            sensor37 = ColorDistanceSensor(xport)
            print(sensor37.color(), "Dist:", sensor37.distance())
        elif device.ID == 38:
            print("BOOST Interactive Motor  \t\t\tresponse: ")
#            motor38 = Motor(xport)
#            print(motor38.speed)
        elif device.ID == 39:
            print("BOOST Move Hub Motor     \t\t\tresponse: ", end="")
            motor39 = Motor(xport)
            print(motor39.speed)
# 40 (0x28) BOOST Move Hub accelerometer (tilt sensor)
        elif device.ID == 41:
            print("DUPLO Train hub motor    \t\t\tresponse: ", end="")
            motor41 = Motor(xport)
            print(motor41.speed)
        elif device.ID == 42:
            print("DUPLO Train hub beeper")
        elif device.ID == 43:
            print("DUPLO Train hub color sensor")
        elif device.ID == 44:
            print("DUPLO Train hub speed")
        elif device.ID == 46:
            print("Technic Control+ Large Motor    ")
#            motor46 = Motor(xport)
#            print(motor46.speed)
        elif device.ID == 47:
            print("Technic Control+ XL Motor ")
#            motor47 = Motor(xport)
#            print(motor47.speed)
        elif device.ID == 48:
            print("SPIKE Prime Medium Motor  \tresponse: ", end="")
            motor48 = Motor(xport)
            print(motor48.speed)
        elif device.ID == 49:
            print("SPIKE Prime Large Motor  \tresponse: ", end="")
            motor49 = Motor(xport)
            print(motor49.speed())
        elif device.ID == 61:
            print("SPIKE Prime Color Sensor \t\t\tresponse: ", end="")
            color61 = ColorSensor(xport)
            print("Color:", color61.color())
        elif device.ID == 62:
            print("SPIKE Prime Ultrasonic/Distance Sensor \tresponse: ", end="")
            dist62 = UltrasonicSensor(xport)
            dist62.lights.on(100)
            wait(100)
            print("distance:", dist62.distance())
            dist62.lights.off()
        elif device.ID == 63:
            print("SPIKE Prime Force Sensor  \t\t\tresponse: ", end="")
            button63 = ForceSensor(xport)
            force = button63.force()
            dist = button63.distance()
            press = button63.pressed()
            touch = button63.touched()
            print("F:", force, "D:", dist, "Prssd:", press, "Tched:", touch)
        else:
            print("Huh I'm another device. on port", xport)

# print(Port.A)
# print(str(Port.A))
# Port.A
# print("+++++++++++++++++++++++++++++++++++++++++")

who_am_i(Port.A)
who_am_i(Port.B)
who_am_i(Port.C)
who_am_i(Port.D)

if EV3_available:
    try:
        who_am_i(Port.S1)
    except Exception as e:
        print("General Exception on S1:", end=" ")
        print(e)

    try:
        who_am_i(Port.S2)
    except Exception as e:
        print("General Exception on S2:", end=" ")
        print(e)

    try:
        who_am_i(Port.S3) # (Port.S3)
    except Exception as e:
        print("General Exception on S3:", end=" ")
        print(e)

    try:
        who_am_i(Port.S4)
    except Exception as e:
        print("General Exception on S4:", end=" ")
        print(e)

```

laurensvalk, 2020-10-10:
Thanks for reporting.

The stop and run button are not implemented/working yet when you start the REPL, which is indeed a bug.

For now, do CTRL + D to get out of the REPL.

BertLindeman, 2020-10-10:
> 
> 
> Thanks for reporting.
> 
> The stop and run button are not implemented/working yet when you start the REPL, which is indeed a bug.
> 
> For now, do CTRL + D to get out of the REPL.

Thanks Laurens.

laurensvalk, 2020-12-07:
This is fixed by https://github.com/pybricks/pybricks-micropython/commit/95d484f7d868bd5b28449ae5c8778f1f81330ff5.

The stop button should now be activated when you enter the REPL, making it easy to stop.

closed 2020-12-07 by laurensvalk

BertLindeman, 2020-12-07:
Confirmed: 
took technichub firmware 
uploaded to Technic hub
Click REPL shows:
```
MicroPython v3.0.0a11-29-g95d484f7 on 2020-12-07; LEGO Technic Hub with STM32L431RC
Type "help()" for more information.
```
And the STOP button is enabled.

Nice, thanks

laurensvalk, 2020-12-08:
Thanks Bert!